arkypita/LaserGRBL #833: Milling

sepaby, 2020-01-07:
Hi, I'm new here. Your program is the best I've ever used. I have a modified BEMBOX laser. I use it as a laser and milling machine so I would like to support the Z-axis and simple milling process settings.

sdudley01, 2020-01-10:
I agree with sepaby...

Are there any plans in the future to add the third axis stepper and a spindle motor control?

I purchased a 3 Axis 0.9J USB GRBL Control Board+Offline Working Remote Hand GRBL Controller LCD Screen for CNC Laser Engraving Milling Machine Wood Router from Amazon.

[](https://www.amazon.com/gp/product/B07D3PRK4M/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

The board has connection for a spindle motor and the hand controller has a button to turn it on/off. 

They sent me a link to downloaded the GBRL software.  After installation I see there is nothing to control a spindle motor (z-axis or speed).  Bummer!

sdudley01, 2020-01-10:
[https://www.amazon.com/gp/product/B07D3PRK4M/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1](url)

arkypita, 2020-01-11:
LaserGRBL is a free software, developed in my spare time, just to be helpful to who need a simple software for laser engraving.

LaserGRBL is not related to ANY seller of hardware. So, if you buy an engraver or a CNC, branded or unbrended via amazon, ebay, bangood etc and they give you a copy of LaserGRBL they are giving you a software that you have not payed for, but that is freely available on internet.

As LaserGRBL is intended for laser (just look at the name) you can't find many options for Z or spindle. However if you look at settings, you could find a checkbox that say "enable z axis support" that give jou jogging control on Z axis.

If you need more control over Z axis (because you have a CNC/mill) maybe you should look for another software

sepaby, 2020-01-17:
Thank you for the type

closed 2020-01-27 by arkypita