arkypita/LaserGRBL #835: Jogging not working

grayjamz, 2020-01-08:
Hello,
I'm a total noob to Laser GRBL 
I installed laser grbl 3.0.24 and hooked it up to my arduino set up.
If I upload a Gcode program it works and moves the stepping motors without any issue.
But If I want to move the axis manual using the jogging control which is not grayed out it does not do anything.
What am I missing,
I would appreciate a nudge in the right direction 
Thanks in advance
Regards

StuartB4, 2020-01-09:
Which version of GRBL do you have? 

[image]

grayjamz, 2020-01-09:
It says Grbl 1.1z

arkypita, 2020-01-09:
1.1z is an unofficial, unsupported, non standard version that I don't know who has created it.

The owner/developer of grbl also does not know where this version come from.

https://github.com/gnea/grbl/issues/775#issuecomment-567160226

arkypita, 2020-01-09:
Can you tell me more about your hardware? Where did you buy it? What control board does it mount?

StuartB4, 2020-01-09:
Maybe a Bachin machine. I've seen a few of their machines with 1.1z.

grayjamz, 2020-01-09:
As I said I'm a total noob.
I bought the hardware on ebay because a channel of the previous  hardware that ran on mach3  stopped responding and I wanted to go USB instead of serial.
I would not be able to tell you more about the hardware. Below is a link to the one I bought on ebay.

https://www.ebay.com/itm/GRBL-Laser-Controller-Board-USB-3-Axis-Stepper-Motor-CNC-Driver-Controller-Board/142913976919?hash=item2146563a57:g:KjIAAOSwhh9byXrS

grayjamz, 2020-01-09:
would I be able to flash it to the official GRBL version?

arkypita, 2020-01-09:
Yes it is possible, but if the producer of that hardware has made some specific customization in this 1.1z firmware there is a few possibility that it doesn't work and you cannot go back

arkypita, 2020-01-09:
Safe way is to buy a fresh new arduino nano (the blue board on the main board), detach the old board and flash the official version on the new one, so you have the original firmware on the original board.

arkypita, 2020-01-09:
You are not the only one with this issue:

https://github.com/gnea/grbl/issues/775#issuecomment-567160226

https://github.com/gnea/grbl/issues/775#issuecomment-567160226

grayjamz, 2020-01-09:
arkypita thank you very much.
Are there fake arduino nanos or can I buy them anywhere without having to worry with counterfeit ones?
If there are counterfeit ones can you recommend where I can buy them?

arkypita, 2020-01-09:
99% of arduino nano you can buy are Chinese clone but they work well.

arkypita, 2020-01-09:
However if you feel lucky you can try flashing directly your board. Mine was only a warning 😉

grayjamz, 2020-01-10:
That is the plan exactly. If it does not work I order a new one. 
Thanks allot again.

closed 2020-12-16 by arkypita