arkypita/LaserGRBL #837: Add the function of simulation processing

lianzaozi, 2020-01-10:
Hello, thank you for providing everyone with such excellent software. I have a suggestion to add the function of simulation processing on the software, so that I can check the trajectory without connecting the hardware, and I can also perform Estimated approximate processing time at a fixed speed, which can improve the usability of the software

arkypita, 2020-01-10:
Use "Emulator"

[image]

lianzaozi, 2020-01-12:
Thank you, great. I think you have plans to add dxf files to your development plan. You can use netdxf to analyze dxf. This library file is great  https://github.com/haplokuon/netDxf

lianzaozi, 2020-01-12:
I want to improve the Chinese translation. What can I do?

lianzaozi, 2020-01-12:
I have made some improvements and additions to the Chinese translation of the software. How can I submit these changed language resource files to you?

Rybec, 2021-05-16:
Two things:

First, this is really obscure.  I would love to be able to do dry runs without having to get into options buried in the configuration files.  Indeed, I didn't even know this feature existed, until I went through the issues here and came across this.  I was planning on putting in a feature request, but I figured I should make sure there isn't already one.  What I would love to see is a button or menu option that just runs a simple emulation showing the path the laser will take.  Even something that just goes through one step at a time, highlighting the path elements associated with each instruction would be sufficient.  That way I can tell exactly what order everything is happening in.

Second, it crashed for me.  I was using the image in this comment: https://github.com/arkypita/LaserGRBL/issues/451#issuecomment-841828807

To reproduce, load that in Inkscape, select the object, and do `Path -> Object to Path`.  Then do `Path -> Break Apart`.  Now, select the two border paths (there are two, which are on top of each other), and do `Object -> Lower to Bottom`.  Now save and load in LaserGRBL.  From there set the Connection Protocol to Emulator, and start the job.  It hangs in the S1000 instruction, and then it pops up an error dialog that says the board stopped responding.  After that, it seems to keep trying, advancing sporadically, but it happens in large jumps, so you really can't see the actual paths it is taking.

vincentBenet, 2024-05-14:
Hi all, Do you have any new about the DXF support integration? It will be very usefull!