arkypita/LaserGRBL #838: Engraving of large png file (380mmx255mm)

jojoplatzer, 2020-01-10:
Hello,
i am using lasergrbl for my RC Hobby. So far it works fine. Yesterday I wanted to engrave some text and a logo. So I made a .png file and imported into lasergrbl.
So far so good. As you can see on the picture 1 (diagonal engraving) my name "Johannes" is very well, but when more "engraving-dots" are coming in the gcode, it starts to offset somehow. I am guessing that, because the last letters of "Piloten" are sharp again. (In the meanwhile I tried only my name as .png and it works perfect). Also a test with horizontal engraving shows "shaking".
Attached the files. If you need anything else, please let me know.
Also interessting, the gcode which was generated by lasergrbl had in the end 23 fails?
Any help is highly appreciated.
Kind Regards Johannes

[image]
[image]
[image]

jojoplatzer, 2020-01-10:
here the .png file.
regards Johannes
[image]

arkypita, 2020-01-10:
All that problem are obviously related to your hardware.

For the problem of "shaking" your hardware is not able to handle the travel speed (or the acceleration/deceleration speed) you have set.

You can try to lower the maximum speeds that you have in your configuration OR you can try to make some improvements to your hardware such as tighten the belts a little or increase the current regulation of the motors.

[image]

> Also interessting, the gcode which was generated by lasergrbl had in the end 23 fails?

LaserGRBL always generate valid GCODE, but sometime grbl report some errors because the data sent via USB are disturbed by electrical noise and received different from the grbl/arduino system.

This reveal your cable is not a good one or your arduino is a poor quality copy.
Please read FAQ: http://lasergrbl.com/faq/ for some hint how to solve this issues

jojoplatzer, 2020-01-10:
Many thanks for your fast answer.

“Bad Arduino” seems to be the correct answer. As mentioned I did not have any problems. Also engraving smaller things works fine.

I “cut” the .png file into 3 pieces and attached you find my name fully clear. Engraving travel needed maximum 5cm.

Yes, in the beginning I did have some “shaking” related to hardware shimmy bearing. That should be not a problem anymore. Therefore I thought related to the “big” travel ways the gcode is somehow wrong.

Anyway, many thanks.

Regards

Johannes

[image]<https://user-images.githubusercontent.com/8782035/72143468-0d091a00-3397-11ea-976e-07425c471977.png>

arkypita, 2020-01-10:
I am pretty sure that the problem is not into generated gcode. I have seen huge machines (bigger than 1m x 1m) working with lasergrbl and producing perfeclty smooth results.

Also pay attention to the feet / support points of your frame: if they are made of smooth plastic or metal they could slip and move due to the hits given by the motor deceleration.

closed 2020-01-27 by arkypita