arkypita/LaserGRBL #839: It does not move

Flamman1soy, 2020-01-11:
I bought a Laser CNC2-6550 (https://www.amazon.es/gp/product/B07TMF3TTN/ref=ppx_yo_dt_b_asin_image_o00_s00?ie=UTF8&psc=1) and use the Laser GRBL software. I am not sure, but from what I have investigated, I think it has an Arduino Nano.

I don't know why it has been deconfigured and stopped working properly, it doesn't move. The laser turns on, but does not move. If I enter the configuration I see that all the values have been changed.

Could someone provide me with the vlaores to put in each line or some file to load the data?

StuartB4, 2020-01-11:
Can you post a picture / screenshot of the settings so I can have a look.

The default settings for GRBL 1.1f are here, but you will have to adjust them to suit your machine. 
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

It looks like it has the chip they use on the Nano boards, it's the square one with the pins on all 4 sides. It should say something like Atmel328 or similar on it. That seems to be the standard chip on a lot of different boards by different manufacturers.

StuartB4, 2020-01-11:
Also, did you know you can not use the USB and the Offline Controller at the same time.
You should unplug 1 before you use the other.

Flamman1soy, 2020-01-11:
Thanks for answering.
I only use USB. The offline controller does not use it. I have put the data of the web that you have told me. I have had to change the $ 3 to 4 which is the one that suits me and is already moving. What else should I change to leave the CNC well adjusted?

On the chip it looks like it puts Atmel Mega328P U-TH 354730 1935WBK

StuartB4, 2020-01-11:
Change $32 to 1 to enable laser mode.
Change $130 and 131 to the exact size you can use. 
For example, if your X axis can engrave 350mm change $130 to 350. Then do the same for $131 for the Y axis.
Then you can use Continuous jog if you set it in LaserGRBL settings.
That way when you press and hold a jogging button it will jog until you let go, or until it reaches the numbers you set in$130 and $131.  It stops you going over the limit.

StuartB4, 2020-01-11:
You can change $110 and 111 to a higher number to allow you to engrave at a faster rate. Any number up to the limit of your machine before it starts to loose steps or running incorrectly.
I set mine to 12000 but it rarely goes that fast.

You can also change  $120 and $121 to a higher number, that will make it accelerate faster to the point where it starts to engrave and when it moves from one part of an engraving to another,  or if you jog.

Flamman1soy, 2020-01-11:
So is it well configured or would you change anything?
[image]

The 130 and 131 does not work for me, but nothing happens: D

Flamman1soy, 2020-01-12:
Now I notice that it is slower and also does not cut. Yesterday I was cutting 2 mm MDF without any problem. Now he barely punches it.

StuartB4, 2020-01-12:
I wrote what to change in my last reply.
Do you know how to change and store the settings.

If not this will help.
Start LaserGRBL and click Connect. It will only work if you are connected.
Goto Grbl menu and select Grbl Configuration.
Select the Value box for the $ you want to change and change the value.
Press Enter and move to the next one you want to change. 
When you have changed them all click on the Write button and then Close.

If the size is wrong you need to change $100 and $101 step/mm for X and Y Axis. 
Change $100 to 80 and jog the machine 40 mm's. Measure how much it actually moved. 
If it only moved 20 mm change $100 to 160.
Do the same with the Y axis.

Change $32 to 1 to enable laser mode.
Change $130 and 131 to the exact size you can engrave.
Change $110 and 111 to a higher number to allow you to engrave at a faster rate. 
Change $120 and $121 to a higher number, that will make it accelerate faster to the point where it starts to engrave and when it moves from one part of an engraving to another, or if you jog.

Flamman1soy, 2020-01-12:
I changed everything and saved the changes. I don't know why, but now it doesn't cut. Record perfectly. Two days ago I was cutting several pieces of different sizes with 17 passes, but now barely marks the wood. I put the power to the maximum that allows me (1000), the speed between 800 and 1000 which is as I put it before, the laser point is correct but not short.

StuartB4, 2020-01-12:
Have you tried changing the M4 for M3 in Target image settings.

[image]

Flamman1soy, 2020-01-12:
I use the LaserGRBL software and I can only select in Laser ON: M3 and Laser OFF: M5.

Record perfect but I don't know why you don't want to cut now. He burns a lot and he didn't do that before.

I have also contacted the seller to see what he says. I wait for response in a couple of days.

StuartB4, 2020-01-12:
If you have GRBL 1.1 you should have M3 and M4. 
Do you have PWM enabled in LaserGRBL Settings, on The Grbl Menu.

[image]

StuartB4, 2020-01-12:
Have another look in Grbl Configuration settings. If $32 is not set to 1 M4 will not be there.

Flamman1soy, 2020-01-12:
Now it comes out. The strange thing is that he had never entered there and it is strange that he is deactivated

I'm testing to see if I can get it cut now because when I put on the M4 it doesn't burn that much anymore. I hope that gets the laser focused. We will cross our fingers.

StuartB4, 2020-01-12:
Ok. :+1:

Flamman1soy, 2020-01-12:
Nothing, there is no way. Some parameter is wrong and does not cut. It is as if there is not enough power even when the laser is at 1000. It does not burn so much with the M4 and the laser is more focused, but it is still uncut.

StuartB4, 2020-01-12:
Have you tried cleaning the lens on the laser?
Have you tried any other laser programs?

Flamman1soy, 2020-01-12:
Yes, and with the same result. If tomorrow does not give me a solution, the seller will return it and buy this one. Do you know any that are good and not very expensive?:
https://www.amazon.es/dp/B07P5C662T/?tag=aloge_eu-21

This is how it cut at the beginning and how bad it does now:
[image]
[image]
[image]

StuartB4, 2020-01-12:
What sort of wood is that. It looks like it may be burning the resin in the wood more than the wood.
Looks like too much power or too slow.
Are you sure your laser is focussed as small as possible. 

I've never bought or used one of them machines you gave the link for. But there are a lot of people that have them.
You can find them on Facebook groups. There's loads of help and information on them.
Like this one:
https://www.facebook.com/groups/1843200505896555

Flamman1soy, 2020-01-13:
In the end I made the return because for more configurations that gave me it did not work. I think it was due to burn the laser.
Already when I have the new machine I will put some photos here for you to see.

Thank you very much for your patience and for answering my questions. People like that are nice to meet.

Greetings and thanks again.

StuartB4, 2020-01-13:
Yeah, Plywood is what I call it. :+1: 
Sorry can't help more, running out of ideas now. Maybe the laser is faulty.
Good luck with the new machine, something else to learn.

Happy new year. :)

closed 2020-01-27 by arkypita