arkypita/LaserGRBL #840: Change M03 and M05

branther, 2020-01-12:
Congratulations for the excellent software.
i use a DIY pen plotter. the pen is moved with a small servo as usual.
my first problem is that i want to change the up and down commands M03 and M05, because my pen is lowered with M05 for drawing and lifted with M03S90. How can I swap the commands? Is there an editing option for the input fields?

my 2nd problem refers to the import or opening of files. when opening my computer hangs. only after the 3rd try, i can access my files. i will report this separately.

I use GRBL 1.1f (CoreXY) with an Arduino Nano(clone) and a modified shield.
thanks a lot

arkypita, 2020-01-27:
1) It was asked a lot of times (search in closed issues) but LaserGRBL is designed to generate code in a very hard-coded way to be grbl "laser mode" compatible. Any hardware that needs different code for turn on/off and modulate is difficult to support.

1) sound strange. how big is the file?

branther, 2020-01-28:
Thank you very much for your answer. Sorry for my late Answer.
i will modify the pen head so that it lifts the pen with m05 and lowers the pen for drawing with m3s0x.

To my second question.
It does not matter how big the file is. As soon as I want to open any file, the application hangs up. 
I have to close it (hard) and call it again. On the third try it works.
I do not know if it is a program problem or a problem with my windows10. I thought maybe the problem is known.
thanks, best regards

branther, 2020-01-28:
By the way, I have this problem ONLY with LaserGRBL.
i made a little screendump movie. 
how can I upload it here?

[Aufnahme-1.zip](https://github.com/arkypita/LaserGRBL/files/4122733/Aufnahme-1.zip)

OK, I have zipped the mp4-File

arkypita, 2020-01-28:
The problem was never reported before.
By your screendump look-like the issue is with your OS.

The "Open file dialog" is a standard system dialog given by the system, that LaserGRBL only recall to select a file.

Seem that the hang occurs during navigation to desktop folder.
- Do you have a lot of file on your desktop?
- The issue occurs only navigating to desktop or also to other folders?
- Can you try to [disable thumbnail preview](https://www.tenforums.com/tutorials/18834-enable-disable-thumbnail-previews-file-explorer-windows-10-a.html) for icons?

[image]

branther, 2020-01-29:
I used a fresh windows installation to test it. 
here too, lasergrbl behaves strangely. As soon as I want to open the file, the icons in the background are changed briefly. in this example, even with the program stuck permanently. 
see movie2
[Aufnahme-2.zip](https://github.com/arkypita/LaserGRBL/files/4128330/Aufnahme-2.zip)

A special feature of my Windows installations is that I have all the data, photos, music, desktop ... etc. move to a separate drive. 
maybe that's the problem?
best regards

branther, 2020-01-29:
i have two further remarks.
i have put my desktop back to the original path. Now the desktop is completely empty... No change

I use a 4k ultra-HD display.
I've reset it to standard HD. ...No change.

arkypita, 2020-01-29:
I have reviewed the code of LaserGRBL and it is clean in what he do.

There are a lot of people that report issue with open file dialog in W10 (not related to LaserGRBL) so I think it is an OS issue: https://www.google.it/search?q=w10+hang+open+file+dialog

Please try open "File explorer option" and see if something change when check/uncheck
- [Always show icon, never thumbnail](https://www.tenforums.com/tutorials/18834-enable-disable-thumbnail-previews-file-explorer-windows-10-a.html)
- [Launch folder windows in a separate process](https://winaero.com/blog/how-to-start-file-explorer-in-a-separate-process-in-windows-10/)

closed 2020-12-16 by arkypita