arkypita/LaserGRBL #841: SVG file path optimization code

lianzaozi, 2020-01-12:
You can refer to LaserPewer software for SVG file path optimization. There is code specifically for path optimization. I do n’t know much about the reference value of this algorithm. I ’m just contacted. You can download it and try it yourself
LaserPewer:  https://github.com/hhtat/LaserPewer

lianzaozi, 2020-01-12:
This is the gif animation and svg file I tested
[image]
[image]
[text_SVGFile.zip](https://github.com/arkypita/LaserGRBL/files/4050470/text_SVGFile.zip)

arkypita, 2020-01-27:
Thanks for the hint.

LaserGRBL has it own TSP optimization (www.google.it/search?q=traveling+salesman+problem) alghoritm that is used when you import raster image and use the "vectorize" feature. You can try it by yourself by turn on "optimize travel" and see the difference

[image]

[image]

[image]

For now it work only on raster import and not on SVG. but when I will implement TSP in SVG import i'll use the alghoritm from LaserGRBL because it is many times faster than LaserPewer.