arkypita/LaserGRBL #843: LaserGRBL software cannot read text in SVG file when reading SVG file

lianzaozi, 2020-01-14:
LaserGRBL software cannot read text in SVG file when reading SVG file

LaserGRBL software
[image]

Inkscape software
[image]

StuartB4, 2020-01-14:
Highlight all objects and click Group on the Object menu. Then click Object to path on the Path menu then save svg.

arkypita, 2020-01-27:
> LaserGRBL software cannot read text in SVG file when reading SVG file

Yes, it cannot. Not so easy to implement since "text" is not saved as path in SVG but as text+font+size info, so it is difficult to import.

Just follow @StuartB4 hint to transform "object to path" and then save SVG that now contains path instead of text.

closed 2020-01-27 by arkypita

jomo487, 2022-06-29:
This subject would be a great addition to FAQ