arkypita/LaserGRBL #849: A VinylGBRL that's as good as LaserGBRL

TheHosk, 2020-01-17:
I'm just writing to see if the same guy who developed LaserGBRL could also develop a VinylGBRL so I can use a dragknife on my CNC machine.
I've had exhaustive looks around to find something that is as easy as LaserGBRL to use and yet to find one. They are either too expensive for what they provide or too complex.
I had seen one addin for $9USD for Fusion 360 but did not work that well and others either did not work with GBRL or were to complex for what they were.
Please can someone come up with a simple app that works as good as LaserGBRL. I have to say I would pay a fare fee for it. There would be a good number of people in that same boat.

Regards

Chris

frankvdh, 2020-04-26:
Can you not cut vinyl with a laser?

StuartB4, 2020-04-26:
`Can you not cut vinyl with a laser?`

Yes you can, but it needs to be special laser safe. If it's PVC the gasses it gives offcan be deadly to humans and to the machine.

regpye, 2021-03-21:
Vinyl is PVC polyvinylchloride. When burn it gives off chlorine gas which is a deadly poison and also has a very bad reaction on electronic components and will quickly destroy them.  Only use a drag knife for cutting vinyl, be safe.