arkypita/LaserGRBL #854: Poor output, missing sections on

moorsey, 2020-01-21:
Hi folks, not sure if this is a bug, or my mis-use of the software, very much appreciated if anyone has any ideas.

I have recently been experimenting with greyscale output on my Eleksmaker A3 engraver, previously only doing black and white.

I've tried lots of different settings, but can't seem to get things right.

With PWM enabled in settings, using M3 or M4, I get very poor output, you can see the random patches of missing image (LaserGRBL has converted fine and the line preview on screen looks great).  Have tried different conversion methods also, but can't seem to "fix" it

https://i.imgur.com/3Bow83A.jpg

With PWM disabled, I get a much crisper output, but can't turn down the laser power, so get a burnt output.  Tried increasing the MM/minute value to compensate, but this doesn't seem to make any difference and output is still burnt (tried up to 5000, but then found a GRBL setting which is less then this, so guess this may have over-ridden what I entered?).

https://i.imgur.com/wXgtjQr.jpg

I'm not sure how to find out if PWM is properly supported on my device, it certainty doesn't fail or throw errors and have used it for everything I've done thus far, aside from a few tests as mentioned.  I have also dropped the power from 255 to 200 for some tests, which makes a noticeable difference, so can only assume this means it is supported

My next thought was faulty USB cable, which I will try swapping this week, the randomness of the missing output suggests noise / interference?

Many thanks

moorsey, 2020-01-21:
Just to close this off, this issue was almost certaintly related to inodes / real time file watch problems.

For anyone coming across this, make sure you try all of the following:

Increase memory - https://support.code42.com/CrashPlan/6/Troubleshooting/Adjust_Code42_app_settings_for_memory_usage_with_large_backups
Increase inodes - https://support.code42.com/CrashPlan/4/Troubleshooting/Backups_stall_due_to_too_many_open_files
Increase real time file watch - https://support.code42.com/CrashPlan/6/Troubleshooting/Linux_real-time_file_watching_errors

Not running via docker currently, as re-purposed an older PC to run Crashplan during troubleshooting with Crashplan.  But no doubt these would fix my issues, as they initially continued when everything was re-set on this PC from Docker, got there in the end!

moorsey, 2020-01-21:
sorry, closed wrong issue with multiple open from other projects....

StuartB4, 2020-01-21:
Do you have any pictures of your laser, and how it's connected. 
Show all the connection leads.

moorsey, 2020-01-22:
Hi @StuartB4 , many thanks, is this of any use:

https://i.imgur.com/ueEOwaK.jpg
https://i.imgur.com/7qzQ9gv.jpg

StuartB4, 2020-01-23:
Is the laser socket twisted to the right because of the strain of the lead, or has it always been like it.
I just wondered because if it's happened on its own it could have loosened the Yellow PWM (S) wire connection pin on the board.  If it's a bad joint on the board it could affect the PWM signal.  I would remove the board and check it on the back, it could be just a tiny crack around the solder, you would see it better with a magnifying glass.

moorsey, 2020-01-25:
Thanks @StuartB4 , checked this today and all seems fine, looks like the socket was just a little wonky.  Might still re-solder it this afternoon though, just to be sure

StuartB4, 2020-01-25:
Ok. It's strange because that board should work fine with PWM and from what you posted at the beginning you seem to have everything right it LaserGRBL.
Is the laser the one that came with the machine?

Just out of curiosity, which version of GRBL do you have?
Just start LaserGRBL and click Connect and it will tell you.

[image]

moorsey, 2020-01-25:
Showing as 1.1e - worth upgrading, if that can be done of course!

StuartB4, 2020-01-25:
If you want to do it, the easiest way is using XLoader and the GRBL 1.1h .hex file which is here, just scroll down a bit to first blue link. https://github.com/gnea/grbl/releases

[image]

Then get XLoader from here: http://www.hobbytronics.co.uk/arduino-xloader
The first link does'nt work, use the second one.
[image]

moorsey, 2020-01-26:
OK, fixed.

This was possibly me twiddling that socket, but, I also swapped the USB cable for a high quality one and what a difference!

M3/M5:
[image]
[image]

M4/M5
[image]

Output is now "perfect", wobble from edges gone, nice crisp lines, all well defined.  Maybe a little dark on the second image, which was on full laser power (255), the others were dropped to 200, but overall, looking good.

Similar result from M3 and M4 settings, I prefer the M3 ones I think.  AM I right in saying that M3 achieves "greyscale" with constant laser power, by instead varying the speed of the head

Thanks for the xloader info, will bank that one for the future, much appreciated

Thanks for your help

StuartB4, 2020-01-26:
Wow!  Yeah real difference. 
Nice to see you sorted it out. 
While it's working there's no real need to upgrade, but as you say, you could always do it at a later date.

Good liuck with it :+1:

closed 2020-01-27 by moorsey

arkypita, 2020-01-27:
> If you want to do it, the easiest way is using XLoader and the GRBL 1.1h .hex file which is here, just scroll down a bit to first blue link. https://github.com/gnea/grbl/releases

@StuartB4 
Easy way is to use the new feature "flash grbl firmware" directly from LaserGRBL

[image]

Is there any problem that doesn't make it work?

StuartB4, 2020-01-27:
No problems for me personally, I just forgot you could do it with LaserGRBL, but I did see someone on one of the FB laser groups that was having problems. That was just after you released that version. Maybe it was them not doing it right, I don't know. Seems to work fine though.
I always wonder though, should I actually tell people how to do it, if there is any problems after they've done it then they have to be sorted out as well.  :)

arkypita, 2020-01-27:
Just asking!

For your information:

In LaserGRBL I am calling avrdude.exe command line, same as XLoader does (and Arduino IDE too).

I don't remember where my avrdude executable comes from, but what I see is that there is an huge difference in bitesize from my avrdude.exe (455K) and XLoadere avrdude.exe (1.878K). Also the avrdude included in Arduino IDE has a different size (550K).

And if you download AVRdude GUI [from this site](https://sourceforge.net/projects/avrdudegui/)  you will find a fourth different size.

I don't know if different size means different avrdude version (with different support for atmel chip?) or only mean that they was compiled with different compilers.

[image]

By the way flashing with LaserGRBL is equivalent to call 
avrdude.exe -patmega328p -b[BAUD] -P[COM] -carduino -Uflash:w:"[FILENAME]":i

avrdude does all the job

StuartB4, 2020-01-27:
Thanks for that, interesting.