arkypita/LaserGRBL #857: Fehler an einem Rechner, sonst nicht.

ABBJay, 2020-01-24:
Ich benutze die Laser-Graviermaschine "Ortur Laser Master 15w"von [Gearbest.com](https://www.gearbest.com/laser-engraving-machine/pp_009731781853.html?wid=1433363#goodsDetail).

Mein Problem: an Rechner 1 funktioniert alles, an Rechner 2 Bleibt der Laser nach etwa 1 - 3 min auf der Stelle stehen und die "Code-Ausgabe" steht ebenfalls. Nach kurzer zeit später kommt eine Errormeldung und die Sicherheitsfunktion schaltet den Laser ab.
Wenn ich sehe dass der Laser steht, kann ich auf "Abort" klicken und danach wieder auf "Starten" und die entsprechende Zeile auswählen wo der Laser weitermachen soll, dann arbeitet er wieder für 2-3 Minuten weiter.

Kleinere Dateien oder Auflösungen können auch bei Rechner 2 erstellt werden ohne dass der Prozess nach kurzer Zeit abgebrochen wird.

An beidern Rechnern (PC 1 & PC 2) ist die gleiche Software sowie der gleiche Treiber Installiert.

Ich kann mir nicht erklären warum es bei Rechner 1 funktioniert und bei Rechner 2 immer stehen bleibt?

------------------------
Rechner 1: 
I7-4770K @3,5 GHz
x64
Win 10

Rechner 2:
Intel(R) Core(TM) 2Duo @2,93 GHz
x64
Win 10

An beiden Rechnern wurde USB2.0 und das gleiche Kabel verwendet.

StuartB4, 2020-01-24:
Überprüfen Sie die Energieoptionen in der Windows-Systemsteuerung des Computers. Möglicherweise befindet sich der Computer im Energiesparmodus.

ABBJay, 2020-01-24:
Das habe ich persönlich durch vor dem Rechner sitzen und warten ausgeschlossen.

Weitere Daten:
[Fehlermeldung](https://www.bilder-hochladen.net/files/m04w-1f-227c.png)
[Code-Ausgabe](https://www.bilder-hochladen.net/files/big/m04w-1g-5bd7.png) (Standbild, keine Veränderung)
[Parameter1](https://www.bilder-hochladen.net/files/big/m04w-1h-be77.png)
[Parameter2](https://www.bilder-hochladen.net/files/m04w-1i-12c8.png)

StuartB4, 2020-01-25:
Ich glaube nicht. Ich führe LaserGRBL auf einem 11 Jahre alten Laptop mit 1,6 GHz CPU, 2 Gig Ram und Windows 7 Pro aus und es läuft stundenlang einwandfrei. Es sei denn, Sie haben viele andere Apps gleichzeitig ausgeführt. Ich habe diesen Fehler noch nie gesehen.

StuartB4, 2020-01-25:
Tut mir leid, ich habe einen Fehler gemacht, es ist eine 2 Ghz CPU, nicht 1.6.

ABBJay, 2020-01-25:
Ich habe jetzt auf Rechner 2 Windows neu Installiert, aber das Problem besteht immer noch.

StuartB4, 2020-01-25:
Haben Sie versucht, das neueste Windows .Net Framework zu installieren? Ich weiß, dass LaserGRBL es braucht, um zu funktionieren.

https://dotnet.microsoft.com/download/dotnet-framework

ABBJay, 2020-01-25:
Hab ich überprüft, ist aktuell (.NET Framework 4.8).

StuartB4, 2020-01-25:
Ok. :+1: 

Auf der Website heißt es, dass Sie den Link für "Schwacher Lichtmodus: Der Laserstrahl stoppt automatisch nach 100 Sekunden. Keiner Schwacher Lichtmodus: Basierend auf der Leistungsstufe stoppt der Laserstrahl automatisch innerhalb von 10 bis 60 Sekunden."

Das scheint das Problem zu sein. So steht es in LaserGRBL in dem von Ihnen geposteten Bild.

[image]

Haben Sie hier verschiedene Einstellungen ausprobiert?

[image]

ABBJay, 2020-01-26:
Ich habe alle Einstellungsmöglichkeiten ausprobiert und jeweils mit Rechner 1 verglichen. Das Ergebnis war immer gleich: zwischen 1 und 7 Minuten hängt sich der Laser auf (bei Rechner 2).
Ich habe auch Windows 7 Installiert und die gleichen Tests nochmal durchgeführt, keine Veränderung.

Vielen Dank für deine Zeit und deine Hilfe @StuartB4 !

Ich geb es auf. Werde einen anderen PC besorgen und hoffe das es bei dem neuen dann Funktioniert.

closed 2020-01-26 by ABBJay