arkypita/LaserGRBL #858: Everything is back to front!

Windmill-Keith, 2020-01-26:
Using both Laser GRBL and LiteFire (Engraver Master or whatever came with the kit) all of the images come out back to front! Reversing the X or Y axis on LiteFire don't seem to make any difference and can't seem to find a setting on Laser GRBL.

Am I doing something wrong? Have I wired something up back to front somewhere?

Any advice/guidance would be greatly appreciated!

BrillDesigns, 2020-01-28:
I had the same issue with my machine (from ebay) using Benbox software. 
Also when using the arrows to move the laser it moved in the opposite direction.

On my board there are 4 outputs X, Y1, Y2 and Laser. I swapped the Y1 and Y2 ones around and that seemed to fix it! 

Maybe it'll be the same for you.

Windmill-Keith, 2020-01-28:
Thanks for getting back to me. That has sorted things out. Thank you

closed 2020-12-16 by arkypita