arkypita/LaserGRBL #859: Wrong speed on engraving

Tweety-99, 2020-01-27:
Hi,

i have a problem by engraving with a Alfawise C30 and LaserGRBL: Sometimes the laser runs the wrong speed. That means if I run the same program another time, the speed is not as programmed(e.g. F1000). So the time grows up to 3 or 4 times and it burns very deep! When I turn off the laser and reload the program, it works fine! If it goes the wrong speed, in the upper right corner is displayed "F0000" the whole time.

I'm using Grbl 1.1f

Does anybody have an idea for this problem?

Thanks for help in advance.

arkypita, 2020-01-27:
Strange issue. Did you use only LaserGRBL or did you use other programs too?

Tweety-99, 2020-01-27:
Hello,

thanks for the fast answer ant your help.

I have installed BenBox, but after some problems at the configuration I have stopped this. Why do you ask? Is there a problem with 2 programs? 

Yours,

Ralf

arkypita, 2020-01-30:
BenBox needs to flash its own firmware on the board, that is different from grbl firmware that is need by LaserGRBL. You cannot use both them or you need to flash a different firmware each time you want to use a program or another (and you lose the machine config) so make your choice.

With Grbl 1.1f speed depends and could be limited by grbl configuration and eventually by speed overrides.

Also be sure you have laser mode enabled. Without Laser mode grbl act as a CNC.
A CNC needs to stop after each speed (power) change to let spindle reach the rotation speed.

[image]

[image]

Tweety-99, 2020-01-30:
I think I found the reason for this problem!

It is a "$32 = 0"- command. In my configuration it is set to "1", but I have a customized button to activate the laser for focussing or positioning with low power. It is a 2-state-button, on and off. To activate the laser without any motion, $32 have to be "0" and switching it off needs to "reset" $32 to "1". If now the focus function is activated and some other(e.g. customized Button...) send only "M5", the laser turned off but parameter $32 is still "0" and at the next start it causes this problem. Reloading the program resets the configuration and all works fine.

Is there an opportunity to visualize the state of the buttons? This could be a great help. I will take a look.

Thanks for helping to solve my problem.

arkypita, 2020-01-31:
> Is there an opportunity to visualize the state of the buttons?

It is already in evidence by a orange background (if you put an image with transparent background on it). This info is not persisted if you restart LaserGRBL.

As you have also noticed, there are situations in which a state may be inconsistent because it has changed from another instance of the program, or by manually sending gcodes, or from a different program.

My suggestion is to write the code of the custom buttons so that it is always consistent (for example using a push-button, which always restores the situation when the mouse is raised, instead of a two-state that needs to be re-pressed).

If you want to be sure that $32 is always on you can insert an initialization code at the start of each job to make sure of the correct state.

[image]

closed 2020-12-16 by arkypita