arkypita/LaserGRBL #862: PROBLEM WITH PWM

segio-1122, 2020-01-27:
I have a problem with my LASER (model: ACAN A12) 15W.
Ok First of all the Laser works only in (VECTORIZE!) 

The G-CODE works correctly I observe the line by line that the code is executed as-is, it turns on M3 with the defined value and M5 turn-off the LASER, but when I put it in the "line-to-line tracing" function the LASER does not turn on.

The first thing I did was check the code and type line by line and it responds, and indeed it does, even if I wrote for example "M3 S100" and so on until "M3 S700" as an example I see that the beam of light is variable.

The next thing was to make a measurement with the multimeter and see the voltages that it handles and that I respond as programmed then 0 is 0 volts and 1000 is 5 volts, the variables that are between 0 to 1000 really when measuring are accurate to the calculations, then I realized that I had no problem, so I decided to take an "image.jpg" and send it the laser I see that the PIN "-Z" of the “CNC shield” which is the PIN 11 of my "Arduino ONE the voltage there is correct but the laser stay on OFF. 

[image]

segio-1122, 2020-01-27:
This is my config list:
[image]

closed 2020-01-27 by segio-1122

StuartB4, 2020-02-24:
Did you manage to fix the problem?

segio-1122, 2020-02-24:
No, in fact I still have the problem :(, I tried the solution you provided me 28 days ago to change the m3 to m4, do you remember it? In the other link "https://github.com/arkypita/LaserGRBL/issues/863", but it didn't work, even try another option about changing the frequency to 7 khz, without success yet.

segio-1122, 2020-02-24:
> Did you manage to fix the problem?
No, in fact I still have the problem :(, I tried the solution you provided me 28 days ago to change the m3 to m4, do you remember it? In the other link "https://github.com/arkypita/LaserGRBL/issues/863", but it didn't work, even try another option about changing the frequency to 7 khz, without success yet.

segio-1122, 2020-02-24:
Can you help me StuartB4?? :( do you have any idea?

StuartB4, 2020-02-24:
I assume you have PWM enabled in LaserGRBL settings.
[image]

The only other thing I can think of is try adding another ground wire from the laser driver board ground to one of the ground pins on the cnc shield, any one of the row of pins on the right hand edge of the shield.