arkypita/LaserGRBL #863: PROBLEM WITH PWM DON´T WORKS ON "LINE TO LINE TRACING" ONLY "VECTORIZE!"

segio-1122, 2020-01-27:
I have a problem with my LASER (model: ACAN A12) 15W.
Ok First of all the Laser works only in (VECTORIZE!) 

The G-CODE works correctly I observe the line by line that the code is executed as-is, it turns on M3 with the defined value and M5 turn-off the LASER, but when I put it in the "line-to-line tracing" function the LASER does not turn on.

The first thing I did was check the code and type line by line and it responds, and indeed it does, even if I wrote for example "M3 S100" and so on until "M3 S700" as an example I see that the beam of light is variable.

The next thing was to make a measurement with the multimeter and see the voltages that it handles and that I respond as programmed then 0 is 0 volts and 1000 is 5 volts, the variables that are between 0 to 1000 really when measuring are accurate to the calculations, then I realized that I had no problem, so I decided to take an "image.jpg" and send it the laser I see that the PIN "-Z" of the “CNC shield” which is the PIN 11 of my "Arduino ONE the voltage there is correct but the laser stay on OFF.

[image]

MY CONFIG:

[image]

StuartB4, 2020-01-27:
You should have M4 to turn the laser on in Target Image settings. 

[image]

segio-1122, 2020-01-27:
Hi! StuartB4 imminently I read your post I tried but the problem continues  :(

segio-1122, 2020-01-28:
Ok thanks arkypita

segio-1122, 2020-01-29:
well now I tried another thing, I changed the parameter $32=1 to $32=0, and now happens something new, when I put on "line to line tracing" the LASER move but only is a turn-on in the perimeter, but not in the fill, another thing that I observed is missing the command M4 in the final menu from the image. and the when I return all again $32=0 to $32=1 and put the LASER on "line to line tracing"  the LASER moves but don´t a turn-on in nothing even on the perimeter.

closed 2020-12-16 by arkypita