arkypita/LaserGRBL #864: Sizing image in LaserGRBL

GAMKCMO, 2020-01-28:
I have reading many comments on this and still can't figure it out.  I'm using 3018 pro with Easel & LaserGRBL.  Making Cribbage board.  Using Easel for drilling and want to use LaserGRBL for the labeling.   All the design work is done in InkScape,  in layers that contain holes and the labeling.  They all fit on top of each other perfect and yet I can't get the Laser part to align?

arkypita, 2020-01-28:
In my test LaserGRBL perfectly respect size and position from svg file.

Just be sure that in "Document property" in inkscape you have set millimeters as "document unit" (not only as view unit). Also set no margin, no scaling, no viewbox.

GAMKCMO, 2020-01-28:
Thanks for your help. Should I be importing from LaserGRBL directly from Inkscape?  I think LaserGRBL likes svg and Easel needs png.  Or does it make a difference?

 

Greg

arkypita, 2020-01-28:
> Thanks for your help. Should I be importing from LaserGRBL directly from Inkscape? I think LaserGRBL likes svg and Easel needs png.

LaserGRBL can open both svg and png/jpg

SVG files are loaded "as is" without filling, resizing, or repositioning options, and even without layer managment (engrave/cut all path from the file) so if you want to do two different processing for one layer and another layer you should save 2 different files, one for each layer.

png/jpg raster file are loaded by grbl and processed with a lot of options like grayscale managment, vectorization, filling options etc but - of course - you will lost "geometric precision" of original vector file.

GAMKCMO, 2020-02-10:
Your probably getting tire of me,  I redid everything in Inkscape and saved to 2 different files as you said.  (see Attached).  When I open the Text one up in LaserGRBL  I don’t get any errors ,but I see nothing?  Opening the Hole file up in Easel it does like it.  If I understood correctly this is the best way to keep the dimensions the same.   I may need to get on the short bus?

 

Greg

StuartB4, 2020-02-10:
Hi Greg, there's nothing attached. I don't think it works if you are doing it from your phone.

> When I open the Text one up in LaserGRBL  I don’t get any errors ,but I see nothing? 

Did you select all the text and click Object to path on the Object menu in Inkscape ?

GAMKCMO, 2020-02-11:
Stuart,   OK I have done as you said.  I found Object to Path in the path menu.  I do get the image to appear now,  but it is very small and the sizing options don’t come up as before.   

 

Attached above is the current file I’m working with.   This is from my laptop.

 

Greg

StuartB4, 2020-02-11:
Hi Greg. Your attachement is still not showing up.

GAMKCMO, 2020-02-12:
Stuart,   Don’t know why my attachments won’t come tru.  I am making some progress and have been able to import SVG  in Plain SVG.  Didn’t know there were different kinds of SVG?

                It comes up and appears to be approximate size.   

·         Is there a ZOOM anywhere

·         As a SVG  I’m unable to choose Horizontal and any other options?

·         Would like to start  at a spot other than lower left   (for alignment purposes)

·         Any other web sites I should be reading?

  

Thanks for any help

Greg

StuartB4, 2020-02-12:
Unfortunately these are the only settings you can change when using .svg files. The reason is that the programming routine for using svg was already writted by someone else and not Arkypita the author 
of laserGRBL. It was imported in to it so there are no additions or changes to the original code as for as I know. 
[image]

You can open .svg files with a text editor like Notepad++ which is free. If you know what you are doing and understand it you may be able to edit the file somehow.

GAMKCMO, 2020-02-13:
Stuart, Was able to make some sizing adjustments in Inkscape.  By Jogging the laser position  to different points I wanted to engrave I could make adjustments in Inkscape without actually running the program.  Saving the Inkscape file as “Plain SVG”  made a big difference.

 

Thanks for your help

Greg

 

 <https://user-images.githubusercontent.com/16980026/74380995-91115000-4de2-11ea-9743-b10affa03e09.jpg> rtrt

GAMKCMO, 2020-02-21:
What does the Linear Slide do?  Normally leave all of them alone but was
working on some hard maple and wasn't getting much progress.  Think I
understand power and speed.   Increasing power or slowing speed should
increase the burn I think.

 

Greg

GAMKCMO, 2020-02-24:
I notice a option for using the Z position to engrave deeper.  It refers to
making a button.  So if I make a button does it move Z on every pass or just
move it each time I hit the button.  I'm having a hard time engraving on
hard maple. I have made up to 20 passes at 100% at different speeds.  Only
have 3w so that is probably part of my issue.  Also notice it is doing the
outside and inside of the digits.  Is that because of the digits (font) I
used or is there a selection in the GRBL I missed.  

Still need the below answered.

 

Thanks for your help,  I'm learning slowly with your help!

Greg

 

From: Greg Mansker [mailto:g_mansker@att.net] 
Sent: Friday, February 21, 2020 5:15 PM
To: 'arkypita/LaserGRBL'
Subject: GRBL override linear

 

What does the Linear Slide do?  Normally leave all of them alone but was
working on some hard maple and wasn't getting much progress.  Think I
understand power and speed.   Increasing power or slowing speed should
increase the burn I think.

 

Greg

StuartB4, 2020-02-24:
I assume you have PWM enabled in LaserGRBL settings.
[image]

The only other thing I can think of is try adding another ground wire from the laser driver board to one of the ground pins on the cnc shield, any one of the row of pins on the right hand edge of the shield.

I don't have a Z axis so have never used the Z axis button. But I would think you would have to do 1 pass then use the button to drop the laser a tiny amount that you set in the button's Gcode box.

GAMKCMO, 2020-02-24:
I check the PWM and it is checked.   I can add gnd from audrino to laser driver.  Where would it go on the driver,  where the other gnd is.   It is burning the digits just very lightly.   

 

What is the linear over ride for and I’m I correct in the slower the speed the more burn per pass?

 

 <https://user-images.githubusercontent.com/16980026/75181759-e2043b00-5736-11ea-85bf-4d13d2730867.jpg> tytytt

StuartB4, 2020-02-24:
Sorry GAMKCMO. The bit about the extra ground wire was meant for segio-1122. I replied to your post by mistake.

Don't know what the linear slider is for, but I never use them.

GAMKCMO, 2020-02-24:
How can we find the answers?

gmmanonymus111, 2020-02-25:
@GAMKCMO For the Z axis problem, LaserGRBL is now able to lower the Z axis after every pass. You can set it (or use custom commands) at _LaserGRBL->GRBL->Settings->GCode->Multiple Passes_ menu. Also, you can add a minimal Z control (up and down arrows) by enabling _LaserGRBL->GRBL->Settings->Jog control->Show Z up/down control_
I have a 1.5W NEJE laser, and I was able to engrave on plywood, bamboo, some kind of pressed board (?), not sure if any of them was hard wood, though.
Lastly, LaserGRBL ignores (or does not recognise) the default Inkscape filling, and that could explain why does it engrave only the inside/outside of the digits. See #873

GAMKCMO, 2020-02-25:
Alex,  Thanks for the info  I’ll check it out.  What about the linear slide,  what does that do?

 

Thanks

Greg

gmmanonymus111, 2020-02-25:
Sorry, I don't quite get where is that 'Linear Slide'. Could you tell me more where to find it?

GAMKCMO, 2020-02-25:
At the bottom where speed and power are

StuartB4, 2020-02-25:
@gmmanonymus111 
[image]

GAMKCMO, 2020-02-25:
3018 pro 3 watt

I think I must be having a laser problem.   I gone back to some earlier pieces I practiced on and it not making a scratch on anything.    The focus seems about the same and it fires as it goes through the motions but no burn?   The only thing I have done was switch my laser button so it would toggle on and Off for the low power when I focus.  Unscrewed the focus lens and all looks clean.  Don’t know where to go from here.  

 

Stuart show me where the linear slide is which I knew,  want to know it’s function?

 

Greg

StuartB4, 2020-02-25:
Sorry GAMKCMO. I was'nt showing you, I was showing gmmanonymus111 because he asked earlier
where it was when he wrote this:

> Sorry, I don't quite get where is that 'Linear Slide'. Could you tell me more where to find it?

I just thought it would be easier for him to find it if he saw a picture.

GAMKCMO, 2020-02-26:
I think I resolved the below condition.  I was unaware that the laser should be at a certain height above the project.  About 50 /55mm.  I found this from a face book group. I guess till now I was just lucky to be in that range.  By raising to that height I was able to get a better  focus  and engraving away.  Plan to make a setup block so when I move from routing to laser I can get it set correct.

Still don’t what the linear slide does?

 

Greg

 

From: Greg Mansker [mailto:g_mansker@att.net] Sent: Tuesday, February 25, 2020 3:41 PM
To: 'arkypita/LaserGRBL'
Subject: RE: [arkypita/LaserGRBL] Sizing image in LaserGRBL (#864)

 

3018 pro 3 watt

I think I must be having a laser problem.   I gone back to some earlier pieces I practiced on and it not making a scratch on anything.    The focus seems about the same and it fires as it goes through the motions but no burn?   The only thing I have done was switch my laser button so it would toggle on and Off for the low power when I focus.  Unscrewed the focus lens and all looks clean.  Don’t know where to go from here.  

 

Stuart show me where the linear slide is which I knew,  want to know it’s function?

 

Greg

gmmanonymus111, 2020-02-26:
Thank you @StuartB4 & @GAMKCMO, I found it. Since I could not make it to save the values I set, I assume it should modify working speeds of the machine, in real-time. From grbl wiki:

> - Feed Overrides
> 
>   - Immediately alters the feed override value. An active feed motion is altered within tens of milliseconds.
>   - Does not alter rapid rates, which include G0, G28, and G30, or jog motions.
>   - Feed override value can not be 10% or greater than 200%.
>   - If feed override value does not change, the command is ignored.
>   - Feed override range and increments may be changed in config.h.
>   - The commands are:
>     - `0x90` : Set 100% of programmed rate.
>     - `0x91` : Increase 10%
>     - `0x92` : Decrease 10%
>     - `0x93` : Increase 1%
>     - `0x94` : Decrease 1%

More [here](https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#extended-ascii-realtime-command-descriptions)

GAMKCMO, 2020-02-26:
Thank for the info,  very interesting read,  didn’t see anything referring to the Linear question.

 

Thanks 

Greg

 

PS  what are you doing up at 2:29 AM

gmmanonymus111, 2020-02-26:
As it says, it is a feed override, which means the speed of the job will be modified by this value, e.g. You have a feed rate of 150 while engraving, but it is too light, you can modify the override value to decrease it to 100 or 75 without altering the GCode. Jogging is not modified by the overrides, if you want to test it, you have to start a short job, like moving in a circle or rectangle.

Side note: The GRBL emulator does not work with these overrides, but the real one (on a MCU) let me save it's values. 

Also, here's only 18:12 or 6:12PM ^^

GAMKCMO, 2020-02-26:
Thanks again,   still no mention of the 3rd option that is LINEAR and what it does? What is (real one (on a MCU). 

 

Greg

gmmanonymus111, 2020-02-27:
I meant an Arduino with GRBL firmware as a real one (MCU stands for MicroController Unit)
For the linear part:
Power slider modifies S (power/spindle speed) commands
Linear slider modifies G1 (during work) commands
Rapid slider modifies G0 (travel) commands

GAMKCMO, 2020-02-27:
Alex,   I found this info and I think I finally understand what the Linear does.  Guess I’m a little slow on terms.

            G0 = speed travel when not cutting or engraving

            G1=speed when  cutting or engraving

Thanks for your patience

Greg

·          

·         G0 is rapid move. It moves the machine to the given coordinates, with the expectation that no machining takes place during the move (tool not in contact with the stock). Unless the mode is changed by other G-code commands, the coordinates are absolute: G0 X10 moves to X=10, G0 X10 again does nothing because you're already at X=10. If coordinates for two or more axes are given, the machine moves in a straight line to the specified point. Starting from X=0, Y=0, the command G0 X10 Y10 moves diagonally to X=10, Y=10. Even if the axes have different maximum speeds and accelerations, the machine still moves them in a coordinated fashion, so that the move is linear. Only the axes specified move; the others do not change position.

·         G1 is linear move. It moves the machine to the given coordinates, with the expectation that the tool would be cutting. It's exactly like G0, except it has an extra parameter, F, that gives the feed rate (the speed at which to move). The speed is expressed in units per minute (mm per minute or inches per minute, depending on mode -- default mm/min). So, with the machine at X=0, Y=0, the command G1 X100 Y100 F200 moves it diagonally to X=100, Y=100, a distance of about 141 mm, at 200 mm/min, so it takes about 43 seconds to get there.

·         G2 and G3 are arc moves (clockwise and counter-clockwise), and they're described very well here. I would use only the I J form and not even bother with the R form. <http://www.shapeoko.com/forum/viewtopic.php?f=3&t=4008&p=30036> [2]

closed 2020-12-16 by arkypita