arkypita/LaserGRBL #865: Laser not working

SxnBlackMamma, 2020-01-28:
Hi, I've bought an Alfawise C30, mounted it and installed LaserGRBL. The machine seems to connect correctly, the jogging works, but the light of the laser doesn't get out. Of course the power supply is connected, the cooling system works. I've tried the command $32=1 but nothing change, if I run a job, the machine moves but no laser at all. Can someone help me?

gmmanonymus111, 2020-01-30:
Have you tried turning it on with a command like `G1M3S1` (this should work while $32 is set to 1), otherwise `M3S1` should be enough (Note: you can use M4 insted of M3, those commands turn the laser on, S1 sets it to low power mode while S255 sets it to maximum power with default GRBL settings)

SxnBlackMamma, 2020-01-31:
Alex, you have a place in my heart. If I use G1M3S1 I have a message: undefined feed rate. If I use just M3S1 the program is happy but nothing happens... I think that the laser module is defective. It's so sad to see that every youtuber who tried this machine had no problem while I am cussing since monday... By the way, I have two pieces that I don't know where to place. They were not mentioned in the instruction sheet. Can you help me with this question?
[image]

gmmanonymus111, 2020-01-31:
Sorry, I think I forgot to specify a feed rate, the command should be `G1F1M3S1` 

Make sure the laser haven't been dropped, received too much current or heat. If the package was damaged, it most likely the laser is damaged, too. They are very sensitive, unfortunately.
If everithing seems all right, you should check the wiring on the laser, especially the driver. With a multimeter, you could measure the output voltage where the laser's wires are connected to the driver board. It should be ~5V. (Just to make sure that part works). Also make sure the polarity is right, though the driver shouldn't work with reversed polarity. If it's working, then it is most likely your laser is at fault. (To make sure, you'd need to disassemble the whole laser module, and since it is pressed tight like the most of the modules, I wouldn't recommend it).
Most lasers come with a guarantee, I'd try to contact the store you bought from.

Also, those heatsinks are for your stepper drivers, I think. At least they are very similar to the ones that came with my [DRV8825 stepper driver](https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQfygVphUUJ4cgsMxJWQ67ia-Gc9JTJ4RY9jcNKLKTqBrzxFfrU&s)

closed 2020-12-16 by arkypita