arkypita/LaserGRBL #866: SainSmart Genmitsu 3018PRO with 5.5W laser

Geierma, 2020-01-28:
Hello, I NEED HELP with using the 5.5W laser bought from SainSmart. The GRBL Laser software is not controlling the laser's on/off function. I have been using the 3018PRO with the 2.5W laser for about a year now with no issues. I attached and plugged in the 5.5W laser and BAM....always on. The 5.5W laser comes with a second controller module that has an on/off switch. I have the stand alone controller also. With the controller connect (not the computer) and the laser module's bottom ON, i can control the laser's ON/OFF with the controller. But once i unplug the controller and switch to the computer the laser is constant ON. How do i get the software to control the laser? M5 command does not work.

Geierma, 2020-01-28:
I loaded a .nc to the SD card in my controller. Turned the laser module ON. Used the controller to turn the laser off. Then started the .nc file. The laser turned on at the home position, traversed to start point and completed the image, stayed ON while traversing back to home position and remained laser ON.

Geierma, 2020-01-30:
I plugged in my 2.5W laser and everything worked as designed. Also i used M3 S255 to turn on the laser and M5 to turn it off. Swapping to the 5.5W laser, same as before. Constant on whenever the ON/OFF switch on the external module is on. The software will not control the laser. M5 did not work as it did with the 2.5W laser.

Geierma, 2020-01-30:
Update #4.........I ran a image that would cause the laser to turn OFF and traverse to a new point and turn ON. Nothing!!! the Laser remains on the entire time. PLEASE HELP!!!!

gmmanonymus111, 2020-01-30:
Hy, since I couldn't find your actual lasers, I can only guess. Did you buy the correct one? I've seen 3018PRO and 3018PROVer CNC machines and their seperate laser accessories, and they are not compatible with the other machine's accessories (at least that's what the description says). There might be a problem with the TTL frequiency, or somehow the wires are in the wrong order.

> With the controller connect (not the computer) and the laser module's bottom ON, i can control the laser's ON/OFF with the controller. But once i unplug the controller and switch to the computer the laser is constant ON.

I would check the wiring first, as your laser is constantly on.

closed 2020-12-16 by arkypita