arkypita/LaserGRBL #869: Custom On/Off stopped working

wmadkins23, 2020-01-29:
I created the custom button Laser on/off
M3 S30            M5
Always on

Worked for a few days then stopped working. I did change $32 laser mode =1 
Did not make any other changes?  Anything else I need to look at. This is an important button to focus my laser (15w) before starting a project.

StuartB4, 2020-01-29:
This is my button to switch laser on low power for focussing.
It also moves the laser away from home 50 mm's first.
Then returns home afterwards.
Just makes it easier for me to get to the focus ring.

Button type: TwoStateButton

Gcode box 1

$32=0
G0 X50 Y50
M4 S3

Gcode box 2

$32=1
M5
G0 X0 Y0

wmadkins23, 2020-01-29:
I will try it thx

wmadkins23, 2020-01-29:
It works. Thanks

StuartB4, 2020-01-29:
Ok. Thanks for letting me know. :+1:

closed 2020-02-05 by arkypita