arkypita/LaserGRBL #872: Positioning of imported SVG files

Shroomo, 2020-02-03:
Hi guys. I'm pretty new to this world, learning as I go. I've had no issues with Raster importing, the image always lines up with the bottom left corner at 0,0. However when I import an SVG from Inkscape, LaserGRBL positions it in seemingly random places. I have checked the file on Inkscape to make sure there are no 'invisible' things off the page anywhere. I'm working on a small scale and precision of the placement is crucial.

Thanks a lot. Kris

[image]

arkypita, 2020-02-03:
link to the svg file?

Shroomo, 2020-02-03:
Sorry, here
[Pent with vines.zip](https://github.com/arkypita/LaserGRBL/files/4149026/Pent.with.vines.zip)

arkypita, 2020-02-03:
Loading your file does the same to me.

I simply create a new file (default A4 file with unit in mm), copied your artwork (selecting with mouse, ctrl-c) pasted to this new file (ctrl-v), aligned to page margin bottom-left and saved.

Now LaserGRBL load it correctly.
Also if I resize page to 7x74 it load correctly.

Try yourself and let me know.

Don't know why. Something in original file?

Shroomo, 2020-02-03:
I followed those steps and it did load up in lasergrbl fine. I tried the same with a different design (same size) but it didn't work. It must mean it's a problem with my use of Inkscape rather than a Lasergrbl issue. I'll keep trying and maybe ask on an Inkscape forum!
Thanks a lot for your quick help. And thanks for a nice piece of software which makes it easy for noobs like me to get creating. Cheers, Kris

arkypita, 2020-02-03:
I do not know. Unfortunately I didn't write the part of the code that imports the SVG (I used a third part code) and I don't have any knowledge of how SVG encoding works, to get my hands on that code.

The only things I know is that SVG has a lot of variations in its format.

closed 2020-02-03 by Shroomo

StuartB4, 2020-02-04:
This might help. Just been playing with Inkscape.

I started with an A4 size document, added a few shapes and some text and saved it as a .svg file.
When I load it in to LaserGRBL I get this, which is what usually happens with SVG files.

[image]

The I looked at the document properties in Inkscape and found this.

[image]

Clicking on the "+" sign I get this.

[image]

Clicking on it I get this.

[image]

After saving the file again and loading in to LaserGRBL I get this without the X and Y Offsets.

[image]

You could start off with any size document and resize it this way so it makes the document just the size you want upto the edges of the objects in it. Then use the Margin adjusters to give an offset for X and Y if you need it.

[image]

Like this.
[image]

Shroomo, 2020-02-04:
Hi StuartB4, Thanks for your help and suggestion. I tried following your steps but it didn't work for me. LaserGRBL still positions it at -290, 224. 
[image]

For now I think I'll export the images as PNG and use LaserGRBL to produce a vector path or Raster it.
Thank you

StuartB4, 2020-02-04:
Just tried it with your file and it works. 
I loaded the file. Clicked Document properties on the file menu.
Clicked the + sign and then the Resize page to drawing or selection button.
Select all the objects and group them with Group on the Object menu, then Object to path on the Path menu.
Then save as a .svg file and load in to LaserGRBL.

[image]

arkypita, 2020-02-04:
Hi @StuartB4 
I follow your procedure step by step and does not work for me too.
Inkscape 0.92.3 (2405546, 2018-03-11)

arkypita, 2020-02-04:
Only copy and paste to a new document works for me.

StuartB4, 2020-02-04:
Strange. 
I'm using version 0.91 but I would think it should work in different versions.

StuartB4, 2020-02-05:
Just uploaded a video of how I do it.
https://youtu.be/nYyP6aLuMZY

arkypita, 2020-02-05:
https://youtu.be/AbWwn41vpVU

StuartB4, 2020-02-05:
Wow, that is really weird. 
I wonder if it's anything to do with Windows 10, I use 7 Pro,  or the version of inkscape.

StuartB4, 2020-02-05:
I just installed inkscape 0.92  and it's doing the same as it is for you. Not working.
Tried with 0.91 again and it works.
Need to find the difference in the 2 inkscape versions.

Shroomo, 2020-02-07:
After watching a few Inkscape tutorials and reading comments, it seems there are a few issues/bugs with Inkscape .92. So I've installed .91 and now positioning is working fine.
Thank you StuartB4 for your investigations. Hopefully the Inkscape devs will get this sorted soon.
Thanks guys :)

StuartB4, 2020-02-07:
Thats good. Glad you got it sorted. :+1: