arkypita/LaserGRBL #873: SVG Filling

arkypita, 2020-02-05:
Hint for who works with .svg file

LaserGRBL does not create filling for path. 
If you need to create filled region you can do that directly in inkscape (or other favourite svg editor).
I suggest is to use EggBot extension that has a good feature to fill region.

https://wiki.evilmadscientist.com/Hatch_fill

[image]

arkypita, 2020-02-05:
It also contains a usefull feature to render text with single-line path.

[image]

This allows, instead of transforming text into a path, which creates double borders (internal and external), to create very thin text because they are made with a single laser line.

Mirkocpn, 2020-02-20:
where is possible to donwload eggbot?

arkypita, 2020-02-20:
https://wiki.evilmadscientist.com/Installing_software

Mirkocpn, 2020-02-20:
> https://wiki.evilmadscientist.com/Installing_software

thanks

arkypita, 2020-02-20:
you are welcome