arkypita/LaserGRBL #874: board with Arduino Uno, paser upgraded from 5,5W to 20W not working

HonzaPa, 2020-02-06:
Hi, pls help!
I have this board:
https://www.ebay.com/itm/2-Axis-GRBL-Control-Panel-Board-For-DIY-Laser-Engraving-Cutting-Machine/323952930725?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2648
If I had connected 2-wire 5,5W Laser module (to output "motor"), it worked.
With upgrade my laser to this 3-wire module:
https://www.ebay.com/itm/5-5W-10W-20W-Laser-Head-Engraving-Module-Fully-Automatic-For-Carving-Machine/352885219090?ssPageName=STRK%3AMEBIDX%3AIT&var=622141687032&_trksid=p2060353.m2749.l2648
it stopped working (no more working stepper motors when module connected. Unconnected-steppers are working). I connected the laser module to output "laser" (wiring is ok, cooler on laser module is working).
Power suply: old computer power suplay 12V/9A.

pls help me find problem, thank a lot!

HonzaPa, 2020-02-07:
Yes, the wiring is ok (I have changed the red and yellow contascts on the board plug), the fan on LED module works.

gmmanonymus111, 2020-02-07:
Hello,
So to make it clear, the problem is the stepper drivers won't work when the 3-wire laser is plugged in, but they are working outsie the board / inside the board with 2-wire laser?
Your laser probably wont turn on by default when ttl/pwm is enabled (or using GRBL with lasermode) for safety reasons, but should work when you start a job (or turned on manually, G1F1S1M3, to turn off M5)
I had similar problem with my CNCShiled for Arduino Uno, the stepper drivers were working outside the shield, but not within the shield. As it turned out, the 12V connector's negative pin was broken out of the PCB, after resoldering it, everything was working as it should be.

StuartB4, 2020-02-07:
You say you upgraded your laser, which one did you get, what power? 5.5, 10 or 20 watt?
You may need to increase the current on the stepper driver boards to account for the laser using more. But if the laser is using more current than it should it could be faulty. 
I know you said you are using a pc power supply (so do I) so there should be enough amps as it is.

HonzaPa, 2020-02-09:
Upgraded to 20W.
I tried this:
PWM cable (yellow one) connected to the Arduino nano board (from LED laser), but power connectors connected direct from Laser to power suply, and it works! Power suply is the same like for boar with Nano)
Do you know where is problem?
Thank you!

StuartB4, 2020-02-09:
Can you post some wiring pictures of how it is all connected when it works and when it does'nt.

closed 2020-12-16 by arkypita