arkypita/LaserGRBL #875: power level adjustment

angwoodworks, 2020-02-08:
i have a 3000mw engraver from universalengravers.com the 300mw blue laser model, the software support through them is absolute garbage, switched over to lasergrbl today, cannot get images to actually burn onto the surface for engraving wood. is there a power level adjustment available.

i was previously using engraver master but it honestly blows, only thing i liked about it is that when i finally got it to do one thing it actually burned dark. 

help please

gmmanonymus111, 2020-02-08:
You can set laser power during image import. Click open (the icon under connect), then select your image.

- For JPG/PNG images, modify setting to your likings, then click next, there should be the laser power adjustment dialog, along with images size settings.
[image]
- For SVG, you can set laser power after importing your file.
[image]

Schwarzi07, 2020-05-20:
You can also lower the maximum spindle speed in the GRBL configuration

closed 2020-12-16 by arkypita