arkypita/LaserGRBL #876: Zero WCO not zero when resetting

cristianodems, 2020-02-12:
I started to realize that when resetting Zero not zero when, it always stays at X: 40.353 Y: -6.647 Z: 0.024
When I start work without a problem, but when I reset or stop during work, I lose the start position and the job.
[image]

arkypita, 2020-02-13:
WCO it is maintained by the hardware, so if you send a reset it is lost.

cristianodems, 2020-02-16:
[image]

Conseguimos!!!
O G10 L20 P1 X0 Y0 Z0 muda o Wco para Wco X0 Y0 Z0.
basta run code after homing, hard reset arduino, run G10 L20 P1 X0 Y0 Z0 muda o Wco para Wco X0 Y0 Z0, restart LaserGrbl.
Muito obrigado!!!

cristianodems, 2020-03-04:
Solution working

closed 2020-03-04 by cristianodems