arkypita/LaserGRBL #877: problem with laser control

patrik156, 2020-02-13:
please can someone help me with my laser. i have problem with laser control, when i set 
M3 S0 i have max power
M3 S1000 is low
how i can change it
i use grbl1.1 , mana SE 3.2 LASER 5,5 w and TTL to PWM modul

patrik156, 2020-02-13:
[image]

patrik156, 2020-02-13:
[image]

arkypita, 2020-02-13:
Hi @patrik156 
In one of your picture I read PWM=0 : ON, PWM=1: OFF.
This is the opposite from the standard.

You can reverse signal using a transistor as digital signal inverter
[image]

Or if you want a software solution you should re-compile and flash grbl to your arduino control board. There is a [macro in grbl config.h](https://github.com/gnea/grbl/blob/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4/grbl/config.h#L211) that allow to invert logic.

patrik156, 2020-02-13:
https://github.com/gnea/grbl/releases/tag/v1.1h.20190825 
i download and uncommented INVERT_SPINDLE_ENABLE_PIN but now i cant  upload i have this message 
Sketch uses 29762 bytes (96%) of program storage space. Maximum is 30720 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
 before this i use grbl laser for upload

arkypita, 2020-02-13:
it is perfectly normal

arkypita, 2020-02-13:
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#compiling-grbl

patrik156, 2020-02-13:
i still have roblem , can you help me 
In file included from C:\Users\PC1\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC1\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC1\Documents\Arduino\libraries\grbl/nuts_bolts.h:56:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\PC1\Desktop\arduino-1.8.11-windows\arduino-1.8.11\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 

In file included from C:\Users\PC1\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC1\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC1\Documents\Arduino\libraries\grbl/nuts_bolts.h:57:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\PC1\Desktop\arduino-1.8.11-windows\arduino-1.8.11\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 

In file included from C:\Users\PC1\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\PC1\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\PC1\Documents\Arduino\libraries\grbl/nuts_bolts.h:61:0: warning: "bit" redefined

 #define bit(n) (1 << n)

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\PC1\Desktop\arduino-1.8.11-windows\arduino-1.8.11\hardware\arduino\avr\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 

Sketch uses 29762 bytes (96%) of program storage space. Maximum is 30720 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.

arkypita, 2020-02-13:
These issue should be opened on the github page of grbl project, not here.
First, however, do a search on old closed issue.... it has already been requested!

https://github.com/gnea/grbl/issues?utf8=✓&q=warning+redefined

patrik156, 2020-02-13:
yes i find it , but now have another problem , i uncomment INVERT_SPINDLE_ENABLE_PIN , but it doesnt help

StuartB4, 2020-02-13:
> Or if you want a software solution you should re-compile and flash grbl to your arduino control board. There is a macro in grbl config.h called INVERT_SPINDLE_ENABLE_PIN that (if uncommented) allow to invert logic inside the control board.

Just had to do that myself. Bought a new 5 watt laser from wish.com but it has the TTL signal reversed on the built in driver board. So when it's powered and the PWM signal from the Arduino is 0 the laser comes on and stays on. I enabled the INVERT_SPINDLE_ENABLE_PIN in config.h and compiled and re-uploaded grbl to the Arduino and it works perfectly.

StuartB4, 2020-02-13:
Patrik, did you compile and upload grbl to the arduino after making the invert spindle change?

patrik156, 2020-02-14:
yes , but still have max power with M3 S0 

[image]

patrik156, 2020-02-14:
and this configuration is ok ?
[image]

patrik156, 2020-03-14:
[image]
i try to connect NPN transistor , but i am not sure if it is correct . please can you check it please ?

StuartB4, 2020-03-14:
The PWM signal should come out between Pin 3 on transistor and the end of the resistor and goto the PWM wire on the laser.
The other end of the resistor should have a voltage going in, 5 volts is enough.
[image]

patrik156, 2020-03-16:
thank you ,now it works ,but i have one more question because  i think that my laser is weak . so i set M3 S1000 and  input to my laser is 5,5V . it is maximum ?

StuartB4, 2020-03-16:
If you set $30 to 1000 in grbl settings then that will be the maximum laser power.  Any number in between that and the minimum setting on $31 (should be set to 0) will be lower power. So if you set $31 to 1000 in grbl and then set the laser power to 500 in Lasergrbl settings when you prepare an image,  then it will be 1/2 power, 250 will be 1/4 power and so on.

closed 2020-12-16 by arkypita