arkypita/LaserGRBL #881: Laser skip / image not aligned

jillionairedesigns, 2020-02-17:
Almost halfway through this burn (5 hours in), the laser seemed to skip in its alignment and then restart with no other issues.  I checked the alignment and tightness of the black rubber tread, everything seems aligned with no issues.  Any input / ideas what happened?  

[image]

arkypita, 2020-02-18:
LaserGRBL always send commands with absolute coordinates, so if some command is received wrong (ie. with wrong coordinates) the next commands will be ok again: no misalignment could be done by the software.

So, in my opinion, your problem must be investigated in the hardware.

Motor power cable with a dancer connection, profile of the guides with irregularities, which may have stopped the run (if the engine is not sufficiently powerful), insufficient or intermittent power supply, something that has slipped between the laser head and the frame (such as the mobile cable that carries the current to the laser) and that has stopped the race etc

jillionairedesigns, 2020-02-18:
Thank you!  I will take a look and see if I can isolate one of those issues.

closed 2020-12-16 by arkypita