arkypita/LaserGRBL #882: LASER NOT WORKING

quotelawrence, 2020-02-17:
so let me explain and possibly someone could point me in a good direction, I wanted to change laser modules so I took an extra module and hooked it up to my 3018 chinese laser/mechanical engraver, but it was not working, i noticed it said grbl update, and I thought ok, so I hit the update button a message said that this could damage the grbl but I thought why would an update damage it, so I said yes, what happened nothing I cannot get the laser function to work now for either module, I have grbl 1.1h and it has been working correctly but after the update X & Y work but the laser is not turning on and I was considering, removing the entire program and reload the entire program to see if that could correct the issue, any suggestion or thoughts on how to role back the update

quotelawrence, 2020-02-18:
I figured out I would use an extra grbl 1.1f board and it is working again but the image I am writing is reversed so I am at a loss for info on how or what to change to get the image to be the same as the monitor, the monitor looks great what is burnt is reversed and I do not know which setting to change to cause this to be corrected could someone help

StuartB4, 2020-02-18:
Start LaserGRBL and click Connect with the the board that works. Then go to Grbl menu and click Grbl configuration. You can Export the settings from the working board and import them to the new one. 

Or:
Have a look at the list of settings one the working board, especially $2, $3 and $4. Make a note of the values and then change board for the new one and do the same.  If they are different, change the new board settings to match the working one.

quotelawrence, 2020-02-18:
good morning Stuart so just go back connect the old board those settings
were working before I did the
the update, and those were engraving in the right direction copy all
settings from the old board and then
install the new board and adjust the settings to match, great idea I
thought there may have been a setting to
just click to mirror image the subject I will do what you say thanks Stuart

StuartB4, 2020-02-18:
Ok. Let me know how you get on. I have to go out now and won't be here for about 6 hours, but I will check later.

quotelawrence, 2020-02-18:
thanks Stuart I will and because I have 2 lasers I am looking at the one
that is working fine and using those settings

quotelawrence, 2020-02-18:
hi Stuart yes I took the setting from 1.1h and used them on 1.1f and they
did turn it image correct
thanks for the heads up, have a great day

StuartB4, 2020-02-18:
Ok thats good. Happy to help .
You have a great day too. :+1:

quotelawrence, 2020-02-18:
😎

closed 2020-02-20 by arkypita