arkypita/LaserGRBL #885: Engraving per poin Missing

Mirkocpn, 2020-02-20:
Hi, with my laser engraver (Vigotec L7) i used VigoEngraver Software, now i'm using LaslerRGBL and is awesome sofware, but is missimg one of best  option in this, engrave by point,

[image]

Can see that in VigoEngraver can set engraving by poin and select speed, intensity,  how many dot/mm want  and engraving time per poin, this is very good option

think that can improve function this in LaserGRBL software?

thanks

arkypita, 2020-02-20:
LaserGRBL has dithering option. Try it!

Mirkocpn, 2020-02-20:
> LaserGRBL has dithering option. Try it!

i tried , not same definition, inside the egrave looks like laser lost power and not egual each point, problem is that image is dithered but not laser, burning is by line , need to change  line in dot, not continous line on dithered image but different poine each line (in non dithered image), hope that i can explain good

arkypita, 2020-02-20:
> hope that i can explain good

Yes you did.

You are right. LaserGRBL produce dithered image and then engrave it line by line.
Each "point" is done as a line, and multiple point is joined in a single line.

With point engraving you mean that the laser head is firm on a point, then turn on laser for a defined time, then turn off the laser, then it move to the next point, and so on.

Mirkocpn, 2020-02-20:
> Yes you did.
>
> You are right. LaserGRBL produce dithered image and then engrave it line by line.
> Each "point" is done as a line, and multiple point is joined in a single line.
>
> With point engraving [...]

Exacly, like this image result very high definition, always same engrave in each point, you think that can improve this function?

Mirkocpn, 2020-02-20:
I forgot to say thank you for great job with this software

arkypita, 2020-02-21:
> I forgot to say thank you for great job with this software

Thanks

>  You think that can improve this function?

Yeah, I think that could be a nice feature for LaserGRBL.

Can you send me a sample of gcode produced by this tool, and the original image?
I would like to engrave and compare the results.

Mirkocpn, 2020-02-21:
Unfortunately i dint know hw to see it, if you want can download the software for free from “Vigotec.cn” software is vigoengraver v7

StuartB4, 2020-02-21:
@Mirkocpn.  I had the Vigotec software when I bought one of their machines but it was protected with the USB dongle that came with the machine and would'nt work without it.
Have you got a copy that works without the dongle.

StuartB4, 2020-02-21:
Ignore my last post. The version at the top of the list on their site is the one the requires the dongle.

pedrovelandia, 2020-08-10:
Hello, I want to congratulate you for the excellent work of LASERGRBL, I use a VG L7 laser, but unfortunately it presents failures when working with LASERGRBL, that's why I have to work with vigoengraver, but it is not the same, the solution to the problem update the firmware, but this update that fixed the error only works with vigoengraver and not lasergrbl, can anything be done? I will attach the firmware that solved my problem.

pedrovelandia, 2020-08-10:
VigoEngraver_Firmware_V1.3_B20200718

StuartB4, 2020-08-11:
pedrovelandia. 
There is nothing attached.

pedrovelandia, 2020-08-13:
[VigoEngraver_Firmware_V1.3_B20200718.zip](https://github.com/arkypita/LaserGRBL/files/5070054/VigoEngraver_Firmware_V1.3_B20200718.zip)