arkypita/LaserGRBL #886: needed to stop program, now I am hearing alarm and lasergrbl is not working

quotelawrence, 2020-02-20:
so the issue was the heat was way to high so I attempted to alter the program but by stopping it and adjusting I may have caused a major issue, I have tried unplugging engraver, rebooting computer but a sound alarm is squealing and the program is not starting can someone provide a remedy

arkypita, 2020-02-20:
Dear @quotelawrence
I am the developer of LaserGRBL program.

The software LaserGRBL you are using is a free software that anyone can freely download from internet. I developed it in my spare time and it can be used with a huge variety of hardware.

Some vendors supply it with their machines, but I have no relationship with them.

So I don't have the slightest idea of ​​which laser you have, nor where the sound you're talking about comes from. You should contact whoever manufactured or marketed your machine for this request for assistance.

arkypita, 2020-02-20:
However... where does the alarm sound come from?
Your pc? The engraver?

quotelawrence, 2020-02-20:
good morning it is from the engraver, and I ended up pulling the power plug but as soon as I plug it back in the sound comes back on, and I may have to do something else my machine is a cheap chinese 3018 alfawise engraver with laser attachment your software had been running superbly just some of the programs run at a heat which is to hot so the image looks more like a black crater than a picture, so that is what took place I will try reloading the program again thanks

arkypita, 2020-02-21:
The alarm sound from the engraver, is coming from a buzzer/speaker?

quotelawrence, 2020-02-21:
sorry for the delay, after work I came home to fix the problem, once again I plugged everything in and yes the noise was still sounding, but I found it was the motors making the sound I then tried unplugging the motors and reengaging them, but that did nothing. so I went to the parameters settings for some reason whatever I had done had deleted all settings and I just needed to reenter the setting to the grbl $ settings problem solved, thanks for your involvement and prompt responses it really matters,

closed 2020-02-21 by quotelawrence