arkypita/LaserGRBL #887: Exception of type 'System.OutOfMemoryExcption' was thrown, in system with 64GB RAM

Kodaijin, 2020-02-21:
So I was trying to 'vectorize' an image and everytime I try it's throwing up this error: 

> "Exception of type 'System.OutOfMemoryExcption' was thrown"

This seems very unlikely with my system. I'm running 64GB of RAM using a 3900x ;>_>

Attached is two screen shots, one of the settings used, and of the error it's self. The other one is of the image I'm using. 

[image]
[image]
[image]

StuartB4, 2020-02-21:
Same here, I only have 4 gig of ram so I'd sort of expect it. 
It is a highly detailed image, and large file size, maybe there's a limit 
on amount of pixels LaserGRBL can handle. 
Have you tried with a smaller file size, I understand you can only go so small without loosing detail though. 
[image]

Kodaijin, 2020-02-21:
> Have you tried with a smaller file size, I understand you can only go so small without loosing detail though.

Interesting....

So I made several smaller versions just trying to pin down at what size it fails at.
Size
100%: Fail
95%: Fail
94%: Fail
93%: Fail
92%: Fail
91.9%: Fail
91.8%: Fail
91.7%: Fail
**91.6: Pass
91%: Pass
90%: Pass
75%: Pass
50%: Pass
25%: Pass**

The largest pass at 91.6% of the original is sized down from 2292 x 3056 ( 7,004,352 pixels), to 2099 x 2799 (5,875,101 pixels). 

~up shrug~ 

Maybe the software is calling only a limit of 32-bit memory assignment?

arkypita, 2020-02-21:
Even if you have a machine with a large amount of total physical RAM (to which is added a virtual memory created by swap to disk), the amount of memory made available to a process is always a smaller part of it.

Also there is a limit to the size of the individual objects that the code can create (e.g. a bitmap). The maximum size of a CLR Object is 2gb even on 64bit.

https://stackoverflow.com/questions/6107322/memory-limitations-in-a-64-bit-net-application
https://stackoverflow.com/questions/14186256/net-out-of-memory-exception-used-1-3gb-but-have-16gb-installed
https://stackoverflow.com/questions/200348/is-there-a-memory-limit-for-a-single-net-process

[image]

I use to compile LaserGRBL for "AnyCPU" that should produce an executable that could work on 32/64 bit cpu, and the OS select the best mode for the application. However look like it loads as 32bit also on 64bit cpu

[image]

About your pixel computation the issue is much more complex. When LaserGRBL need to create the final output, it scale your original image to a different resolution. This resolution is computed in a complex way just to keep the max resolution possible but without exceeding the memory limit (that I have computed in sperimental way) and should not depend by original image resolution but only to final engraved "target" size.

```
int maxSize = Is32Bit ? 6000 * 7000 : 22000 * 22000; //on 32bit OS we have memory limit - allow Higher value on 64bit
double maxRes = Math.Sqrt((maxSize / (TargetSize.Width * TargetSize.Height))); //limit res if resultimg bmp size is to big
double res = Math.Min(maxRes, SelectedTool == ImageProcessor.Tool.Line2Line || SelectedTool == ImageProcessor.Tool.Dithering ? (double)Quality : 10.0); //use a fixed resolution of 10ppmm for vectorization
double fres = Math.Min(maxRes, FillingQuality);

Size pixelSize = new Size((int)(TargetSize.Width * res), (int)(TargetSize.Height * res));
```

Finally, lasergrbl performs several passes between copies of the image (background cleaning, application of contrast and brightness etc) before the final vectorization. Maybe this is a cumulative effect of this procedure.

I doubt that in previous versions of lasergrbl this was able to handle higher output sizes.
Whould you like to try with this old version? https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.21

arkypita, 2020-02-21:
Edit: I have done this test for you.

[image]

Older 2.8.21 can manage very big images: up to 1.2M x 0.6M and more.
There is something that i have changed that broke something.

Biggest different is that now the OS load the executable as 32bit instead of 64.

I will let you know.

arkypita, 2020-02-21:
The weird things is that after downgrade to 2.8.21 I have reinstalled latest version, and now it load as 64bit application.

Kodaijin, 2020-02-21:
I'll give this a shot and see if I can get it to load as a 64 application following what you did with a downgrade and then upgrade. 

Thank you for responding and taking the time to look at this ♥♥♥

Kodaijin, 2020-02-21:
Yeup, downgrading to 2.8.21 then upgrading to 3.0.24 allowed it to run in 64bit. I saved the g-code and about to get it running ;p 

Let me know if you need me to do any more testing. Where can I buy you a coffee?

arkypita, 2020-02-26:
The issue is very strange and seem to be more a windows issue than a LaserGRBL issue.
The downgrade-upgrade procedure shouldn't have changed anything if not the executable of LaserGRBL itself. 

LaserGRBL.exe is an executable compiled with the "[AnyCPU](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/compiler-options/platform-compiler-option)" flag, which allows it to be run on both 32bit and 64bit machines/os. The operating system chooses the best mode.

For some reason unknown to me, windows may decide to run an executable compiled with the flag "AnyCPU" as 32bit executable, even on 64bit machines. 

> Let me know if you need me to do any more testing. 

Only one information: what is your windows version? (Win10? Win7?)

> Where can I buy you a coffee?

If you like the software and you want to contribute you can give me a coffe via donation: https://www.paypal.com/pools/c/8cQ1Lo4sRA

StuartB4, 2020-02-26:
Hi Arkypita.
I have a 64 bit machine with 4 gig ram and Windows 7 Pro installed.

I tried the downgrade - upgrade option you mentioned but LaserGRBL still installs into Program files (x86) and not Program files where my other 64 bit programs are installed.  So I assume it's still running in 32 bit mode.

I was just wondering if it needs to be completely removed from my system to get it to run in 64 bit mode. As you can see from the image of the Task Manager, it does'nt say which mode it's in when it's running.
[image]

arkypita, 2020-02-26:
> Same here, I only have 4 gig of ram so I'd sort of expect it.

The amount of physically installed RAM memory has nothing to do with the amount of memory that a program can allocate. The operating system provides virtual memory addressing to programs, making a much larger amount of memory available thanks to a [swap file](https://www.computerhope.com/jargon/s/swapfile.htm) on the hard disk.

According to [microsoft docs](https://docs.microsoft.com/en-us/windows/win32/memory/memory-limits-for-windows-releases?redirectedfrom=MSDN#memory-and-address-space-limits) the address space for each 32bit application is 2GB while the address space for each 64bit application is 8TB.

This make LaserGRBL much more limited on x86 system.

> LaserGRBL still installs into Program files (x86) and not Program files where my other 64 bit programs are installed. 

I have read [somewere](https://stackoverflow.com/questions/6305035/why-is-an-any-cpu-application-running-as-x86-on-a-x64-machine) that the installation folder isn't important in the way the OS load the executable. This agrees with my beliefs.

I have added a diagnostic in logger file (? -> open session log) that show if the app is running 32 or 64 bit.

You can now read: 

`------- LaserGRBL v3.1.2 [64bit] START -------`

Between square bracket you can read:
- 64bit: running in 64 bit mode (all fine)
- 32bit: running in 32 bit mode because the OS and/or the CPU is 32 bit (all fine, only an old computer)
- 32bit!: running in 32 bit but LaserGRBL detect we are on a 64bit OS (problem!).

@StuartB4 what do you can read on session log?

arkypita, 2020-02-26:
> As you can see from the image of the Task Manager, it does'nt say which mode it's in when it's running.

You can right-click on table header and add "platform" column.

[image]

StuartB4, 2020-02-26:
Hi Arkypita.

> You can right-click on table header and add "platform" column.

I don't seem to have that option in Task Manager on Windows 7.

But it appears it is running in 64 bit mode so it's ok.
The session log shows it.

> 
> 2/26/2020 17:20:08	Program     	------- LaserGRBL v3.1.2 [64bit] START -------
> 2/26/2020 17:20:11	SetStatus   	Machine status [Disconnected]
> 2/26/2020 17:20:16	SetStatus   	Machine status [Connecting]
> 2/26/2020 17:20:16	OpenCom     	Open COM8 @ 115200 baud Ctrl-X
> 2/26/2020 17:20:16	SendImmediate	Send Immediate Command [0x18]
> 2/26/2020 17:20:16	VersionInfo 	Detected Grbl v1.1h
> 2/26/2020 17:20:16	SetStatus   	Machine status [Idle]

Thanks very much.

Kodaijin, 2020-03-04:
Greeting I've been on holiday. 

Here is the system I am using for making the g-code
Ryzen 3900x
64GB ram
RTX 2080ti
Windows 10 Pro version 1809 (17763.1039)
2x SSD's 1xNVMe 1xHDD
Tons of USB stuff.

arkypita, 2020-03-05:
A great piece of silicon!

I ask because I'm looking for if there is a correlation between OS version and this problem, but I think not. More likely it depends on some 32bit library that the system decides to load, which forces the whole application to load at 32bit

It seems that the problem is very limited, perhaps you are one of the few who has had it and reported.

I will keep it monitored.
Have a nice day

closed 2020-03-05 by arkypita