arkypita/LaserGRBL #890: Laser ttl 20khz

djidai, 2020-02-25:
Hi every one 
I have  driver which has parameters ttl 20 khz. When i try engraving wit  gray tone. I have only 4 colors black, white, and 2 grade of gray. I change frequency in grbl from 1 to 8 khz but i didn't have any changes. Wisely i se what laser change power, but i don't  have  grade in result. Any idea?

djidai, 2020-02-26:
Ohhh i forgot attach the foto of my driver.
[image]

StuartB4, 2020-02-26:
I have the same driver on my laser. When I bought it it said on the website that the PWM signal is inverted on this driver.
[image]

Which means the laser is On at full power when the PWM signal is 0 volts and Off when it's 5 volts.
That is completely opposite to most lasers which are Off with 0 volts and On at full power with 5 volts.
Yours may be the same because I just saw this on the link to the ebay listing where you bought it.

> 5.Intigrate TTL controal slot/0V on and 5V 20kHZ.

If it is you need to invert the PWM signal. 
There are a couple of ways of doing it.
One is to use a transistor/resistor circuit like this one:
https://blog.eccentricworkshop.com/grbl-pwm-inverter-circuit

The one that worked for me was to edit the config.h file in GRBL, then compile it and reupload it to the Arduino.

djidai, 2020-02-26:
but I don’t have any problems, it’s turned on if I don’t do anything and turns on when I give this task

StuartB4, 2020-02-26:
Oh ok, must be 0v =Off and 5v On then like most lasers.
Do you have a picture of your grbl configuration settings?.

djidai, 2020-02-27:
Yes.
[image]
[image]
And I did some changes in files, turned off the z axis and changed the PWM modulation frequency

djidai, 2020-02-27:
and yes I controlled the laser through both m3 and m4, but the result is as I wrote earlier.

StuartB4, 2020-02-27:
Do you have the S-Min and S-Max setting the same in Target Image settings as it is in Grbl $30 and $31.

[image]

[image]

djidai, 2020-02-27:
> I noticed you wrote this earlier.
> 
> > but I don’t have any problems, it’s turned on if I don’t do anything and turns on when I give this task
> 
> Did you mean it's OFF if you don't do anything?

Yes.

djidai, 2020-02-27:
> Do you have the S-Min and S-Max setting the same in Target Image settings as it is in Grbl $30 and $31.
> 
> [image]
> 
> [image]

Yes, I changed these parameters in different ways, but still I do not get an adequate result.

djidai, 2020-02-28:
how did you invert pwm? in a soft?

StuartB4, 2020-02-28:
You have to change it in GRBL before you flash it on to the chip on the controller board.
You can't just change it in software because it gets stored in the chip.
I just changed one line in config.h file, then had to compile grbl and upload it to the chip.

You have to remove the 2 forward slashes // from the start of the line, then save the file, compile it and send it to the chip.
This is what it was before I changed it:

[image]

 and after it looks like this:

[image]

djidai, 2020-02-29:
I tried it, but nothing has changed. I have a default variable spindle.
Sorry you have a DSO Shild or oscilloscope.
you can see the parameters, signal and voltage values ​​after the driver, i.e. what goes on ld?
maybe my driver is faulty.
also what ttl frequency do you use?
[image]
here are my tests for today with different parameters and more.
nothing has changed and at best I have only 4 colors, black, white and about 2 steps of grad.

djidai, 2020-02-29:
[image]
[image]
This is my board and driver.

StuartB4, 2020-02-29:
> Sorry you have a DSO Shild or oscilloscope.

No sorry I don't. I have a digital multimeter but it's faulty and does'nt give proper readings, the digits change to fast to see them.

I use 8kHz frequency, it's actually only 7.8 really.

[image]

So you have 2 power supplies ?
If you use 2 you should connect the ground on the output leads of both of them together.
I use one power supply for everything, it came from an old PC, loads of amps.

Looking at the test piece you posted it does'nt look as bad as I thought it would.
I've done tests myself with similar results, but when engraving an image with all the settings as good as I could get them it turned out ok. A good test is to engrave a photo on something black like slate. Invert the image first so it looks like a negative.  These are on slate with different settings for each one. When you look close they don't actually have loads of different grey scales.

[image]
[image]
[image]
[image]

My system uses a Protoneer CNC shield and a genuine Arduino UNO with GRBL 1.1h, and DRV8825 stepper drivers. I have 2 limit switches one on X and one on Y axis, just used for Homing.

StuartB4, 2020-02-29:
I just noticed you have the 3 pin connector on the board in the picture you posted.
That is usually to connect a 3 wire laser.
+pin=12v,  -pin= ground and S=PWM signal.
Do you have a mutltimeter so you can you check to see if the S pin is connected to pin 11 on the Nano board. If it is then you could use that.

[image]

djidai, 2020-02-29:
> I just noticed you have the 3 pin connector on the board in the picture you posted.
> That is usually to connect a 3 wire laser.
> +pin=12v, -pin= ground and S=PWM signal.
> Do you have a mutltimeter so [...]

my laser is connected.  but I take s and gnd from this connector.
+ - 12 volts I take from an external power supply.
[image]
[image]
[image]
I tried to unify gnd by setting a jumper as in the photo, but it did not give any changes.
also my board has a pwm plug but there is a 12 volt siren wire and i don't use it.
about the multimeter and pwm.  your multimeter is not broken, it just does not understand pwm.  I will use such a cheap device for this.

[image]
You can buy it on togging platforms even a kit for self-assembly.

djidai, 2020-02-29:
I contacted the seller of the driver and here is his answeres. it work like 0V on and 5V off.but in that case, I don’t understand how it works for me.
it works for me 0 off 5 on.
nonsense

StuartB4, 2020-02-29:
Oh right, that is strange. Mine was definately 0 on and 5v off. 
Until I changed the setting in config.h it was on and stayed on as soon as I switched the machine on.
Mine was already connected to a laser because I bought the whole thing together from Wish.com.
It was supposed to be 5 watts but when it arrived it was 7. I also got it half price because I had a discount code.

[image]

[image]

[image]

[image]

[image]

My multimeter is definately faulty, it does random characters on the display when I test AC or DC Voltage and Current. The only thing it's ok with is resistance, diode test and continuity beeper.
Need a new one, had it about 25 years and it's never been calibrated since I got it.

djidai, 2020-02-29:
my friend helped me invert the signal where 0 is 100 and 5 is off.
it does not turn on in idle mode. but now I have a different problem, laser grbl creates the code for normal mode where 0 is off and 5 is on.
but anyway I tried engraving gradient, and it was not successful.
I have no idea what to do with it now.

StuartB4, 2020-02-29:
>  my friend helped me invert the signal where 0 is 100 and 5 is off.

How did your friend invert it?. 
If it's inverted it should be completely opposite, not 100.
It should be 0 and 5, Off and Full on. 

If the people you bought it from say it should be 0=On and 5v=Off then it must be right.
My driver is the same driver so I don't see why they would be different.

djidai, 2020-02-29:
he edited the file cpu_map

StuartB4, 2020-03-01:
Oh right, ok.

djidai, 2020-03-01:
I replaced the driver with the exact one that was in stock.  but contrary to my hopes, nothing has changed.
you have access to your driver, I'm interested in a photograph of the back of the board.

StuartB4, 2020-03-01:
I also asked the seller on ebay how it works and got a completely different answer. I don't think they know how it works. Here's the reply I got. You can see my question at the bottom.

[image]

djidai, 2020-03-01:
apparently it all depends on the weather on Mars.
it blows my brain!!!

StuartB4, 2020-03-01:
Yeah I know what ya mean. I'll get a picture of the undeside of mine in a minute.

StuartB4, 2020-03-01:
[image]

djidai, 2020-03-01:
Thanks for the photo.  it's the same as on my board.

djidai, 2020-03-03:
[image]

I conducted a number of tests, with different speeds and settings.

in fact, I have only 2 colors, these are black and white.
I do not understand what is wrong.

djidai, 2020-03-04:
I foun it on ebay.
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F232923988707
The main interest in this is the description that contains information about the frequency of the supplied signal ttl.
so I’m pretty sure that you need to expand the spectrum of modulated frequencies in grbl.
I spent a series of dough with my driver, grbl and dso.After the signal is processed by the driver, its shape changes, as well as other parameters, the signal becomes weaker.  but in version 1 and 2 kiloherkha it is vaguely reminiscent of a square, in version 7 kHz it has a terrible shape and looks like ^ this.
Any ideas how you can increase the frequency to 20 kilohertz?

or even more?

djidai, 2020-03-08:
I found datashit for the main cheap on the driver.
https://www.rlocman.ru/datasheet/data.html?di=167804&/TPS92641PWP
And look on this 
[image]

HGadotti, 2020-10-12:
Dear friend, did you get the solution for your problem? I would like to buy this driver to assemble a homemade laser. but now I am afraid to use this driver.

mundsen, 2020-10-12:
I ordered one (analog = PWM)

There is 2 different type, one TTL and one analog/PWM

Analog: Intigrate Analogue controal slot----0V to 5V ---weak to strong.Frequncy is 20kHZ.

TTL: Intigrate TTL controal slot/0V on and 5V 20kHZ.

I use a controller with PWM out, so I ordered the analog - hope that is going to work

closed 2020-12-16 by arkypita