arkypita/LaserGRBL #891: Downloaded New Version - Does not Burn!

imagecatcher01, 2020-02-26:
I installed the new version, rebooted compter...it does not burn!  The light for the laser is on, but 
does not burn the wood.

arkypita, 2020-02-27:
I have done some changes in the last version of LaserGRBL, but nothing that can justify this behavior. To be sure that LaserGRBL is commanding to use the max power check your grbl parameters and confirm that are like this:

[image]

Then try to engrave something (a simple logo, with vectorization) with a low border speed and in "Laser Options" set 0 - 255 as min/max.

[image]

Usually the fact that the laser turns on with a weak light that does not burn is a symptom of a "dead" diode. https://github.com/arkypita/LaserGRBL/issues/780

Please note that LaserGRBL does not physically have the possibility of breaking the diode (least of all with an update!).

If the diode stop working it is because it is a faulty one or because it is a poor quality one supercharged with higher powers than it can truly support. 

https://www.youtube.com/watch?v=81Io-ghEIgk

imagecatcher01, 2020-02-27:
Thanks, I will try it again with the new version.  I know it is not a dead
diode because it still burns with the previous version.

Dave Hupke

arkypita, 2020-02-27:
> Thanks, I will try it again with the new version. I know it is not a dead diode because it still burns with the previous version.

Good, I am really happy for you!

In the latest version I have included some changes that have been sent to me by other collaborators, to include support for firmware other than grbl (especially smoothie and marlin). These changes should not have affected Grbl.

Are you using Smootie or Marlin firmware accidentally? Let me know please.

[image]

arkypita, 2020-02-27:
You can check it in your settings.

closed 2020-12-16 by arkypita