arkypita/LaserGRBL #892: Unable to ABORT job

gn3481, 2020-02-27:
Newest version, just updated yesterday.
Unable to abort the job clicking the box. Abort dialog appears, but when I confirm, nothing happens.

StuartB4, 2020-02-27:
It should stop eventually. 
If you do a big image with a high line count and slow speed it takes longer to empty the buffer.
You can see that on the bar under the arrow buttons.

arkypita, 2020-02-28:
Abort is working for me. As @StuartB4 already say the abort job need some seconds to be completed because grbl has to finish the already received commands.

If you think that this is not working for you, please give more info about the issue and how to reproduce, or better record a video of the issue.