arkypita/LaserGRBL #893: Problems with websocket connection

gisep, 2020-02-27:
Hi,
i'm becoming crazy trying to install the "Websocket" sketch to run LaserGRBL via WiFi using a Wemos D1 (ESP8266 chip based), because as the new WiFi connection appears and i connect to this, nothing appears when i digit the adress 192.168.4.1 (as i am offline).

Finding on what could be the reason i noticed these things:
- The adress IPv4 is 192.168.4.2;
- The DNS Server IPv4 is 192.168.4.1;
- The Arduino IDE Serial Monitor stays white all the time;
- I tried to erase the flash memory with esptool.py and put the factory firmware ESP8266_NONOS_SDK, then to run again the sketch, but nothing changed;
- I tried to erase the flash memory with esptool.py, then to run again the sketch, but nothing changed;
- I tried to run another sketch (WiFi repeater) and it runs.

Can someone know what could be the problem?

Giuseppe

gmmanonymus111, 2020-02-27:
Hello, my guess is simply because grbl is not compatible with the wemos d1, so lasergrbl never received the status report (as esp chip has different pinout/clock rate), unless you want to use it as a serial bridge
For me, the websocket was worse compared to the telnet variant, you should try that one, and I recommend using esp-link for that

gisep, 2020-02-28:
Since i'm a noob i didn't mount the GRBL sketch in this board but only the websocket one. 

    That would be really amazing to implement a custom firmware with new added 
    functions like temperature switches or controllers for lights, fans, ... 
    and an embed WiFi connection! This board has got 5 times more power and 3 times 
    more memory than an Arduino UNO, but only one analog INPUT.

So since the chip is the same as the "ESP8266" that arkypita succesfully uses, and since the problem appears before the websocket i don't think the problem is a "websocket compatibility" problem.

By the way i tried to run the telnet sketch and it seems to run correctly!
I just activated telnet on my OS (Windows 10) but i miss the "x" and "y" IP PORT numbers for the address. how can i find them?

Here are some of my telnet options:

> Microsoft Telnet Client
> 
> Il carattere di Escape è 'CTRL++'
> 
> Microsoft Telnet> ?/help
> 
> I comandi possono essere abbreviati. I comandi supportati sono:
> 
> c    - close                 chiude la connessione attiva
> d    - display               visualizza i parametri operativi
> o    - open hostname [port]  connette a un nome host (porta predefinita 23).
> q    - quit                  chiude Telnet
> set  - set                   imposta le opzioni (digitare 'set ?'per
>                              visualizzare l'elenco delle opzioni disponibili)
> sen  - send                  invia stringhe al server
> st   - status                visualizza le informazioni sullo stato
> u    - unset                 rimuove le opzioni (digitare 'unset ?' per
>                              visualizzare l'elenco delle opzioni disponibili)
> ?/h  - help                  visualizza la Guida
> 
> Microsoft Telnet> d
> Il carattere di Escape è 'CTRL++'
> Will auth (autenticazione ntlm)
> Eco locale disattivato
> Modalità nuova riga - Fa in modo che il tasto INVIO invii CR LF
> Modem corrente: Console
> Negozierà il tipo di terminale
> Tipo di terminale preferito: ANSI
> 
> Microsoft Telnet> st
> Non collegato
>
> Microsoft Telnet> o 23
> Connessione a 23...Impossibile aprire una connessione con l'host. sulla porta 23: Connessione non riuscita

gisep, 2020-02-28:
I just realized the port was printed on the serial monitor. 
       ESP8266 Connected!

arkypita, 2020-02-28:
If you want to connect grbl via WiFi you need an arduino uno/nano with grbl firmware loaded on it, and an esp8266 (also wemos d1 should be ok) with websocket or telnet firmware loaded on it.

They should be connected like this.

[image]

Since the esp8266 is only "a bridge" between PC and GRBL, a proper serial TX/RX connection between arduino and esp is mandatory if you want to see some reply because the esp does not reply "directly" to LaserGRBL, it only pass its messages to arduino-grbl and pass back it's reply.

The only exception is that websocket firmware send an "ESP8266 Connected!" message of his own initiative when the connection is opened, even in the absence of the serial connection to grbl.

When you flash websocket/telenet firmware to esp you can see that it create an ah-hoc WiFi network (called ESP8266 if I remember well). You can connect to that network using a laptop or a mobile phone, than go to a browser and go to the address http:\192.168.4.1

You will se a page that allow you to connect the esp to your home WiFi network. Just select and provide WiFi password.

At this point the esp will close the ad-hoc WiFi used for config, and connect to your home network: now the esp is part of your network as any other device and it acquire an IP address in the same range of your PC.

Unfortunately you don't know the IP address it take: you should go to your modem-router management and see what local IP was given to the esp by DHCP managment.

Now you are done! Assuming your modem-router is using a network like 192.168.1.x and your esp take the address 192.168.1.15 you can connect it from LaserGRBL using this url 

`ws://192.168.1.15:81` for websocket
`192.168.1.15:23` for telnet

Personally I prefer telnet because the protocol is more "raw". Websocket add more overhead in communication that is useless and can slow down the data stream.

arkypita, 2020-02-28:
> but i miss the "x" and "y" IP PORT numbers for the address. how can i find them?

Devi accedere al pannello di configurazione del tuo router e cercare una tabella come questa:

[image]

arkypita, 2020-02-28:
Nella tabella "Collegamenti LAN" la colonna "IP" mostra gli IP che il router ha rilasciato ai vari dispositivi di rete, incluso l'ESP8266 (se connesso alla rete). Nella foto che ho trovato in internet hanno nascosto l'ultima cifra ma te l'ho messa giusto come riferimento per capirci.

gisep, 2020-02-29:
Thanks for the help and the suggestions.

I've found it and connected to LaserGRBL, but even if it says "ESP8266 Connected!" it results disconnected, and when i click to connect it waits a bit and then disconnet again.

gmmanonymus111, 2020-02-29:
Is the ESP8266 connected to a GRBL-driven arduino? If not, that is your problem. LaserGRBL waits for a GRBL version string and status report response in order to be connected.
Without a GRBL-riven arduino, neither the version string or the status report response is receive by LaserGRBL, resultin in a time-out.

gisep, 2020-02-29:
Yes, i attached tx with rx and rx with tx between the laserGRBL driven arduino with shield and the wemos D1. I will do other tries in the next days.

gmmanonymus111, 2020-03-01:
Ok, the setup seems all rigth then.
Could you check if it's working with any other telnet client (like putty). with this, it shouldn't disconnect, and when you type something, it should send a response back (either ok, or error:{number}). Also, on connection, it should write the GRBL version and help text.

[image]
[image]

If you managed to get it work, I suggest lowering the 'Threading Mode' to 'slow' in `LaserGRBL->Settings->Threading mode`, as for me, it sometimes froze the ESP8266 chip while it was set to fast or ultra fast.

gisep, 2020-03-01:
I tried to run PuTTY and it reads well the connection.. Opened LaserGRBL and now it runs.
Thanky you for your support @arkypita and @gmmanonymus111!

I just tried the frozening of the ESP putting it to insane. I'm so discovering that the wifi connection is great for some kinds of work, but could be problematic for uses that manage a lot of code, for example to engrave pictures. Setting the thread to fast, insane or slow, i'm seeing that the image is becoming darker than the same with serial connection.

For engraving pictures, the most "omogeneous" and safe result with wifi thread is by putting it to slow (as suggested by gmmanonymus111); the result will be darker than the "serial connection" one, so image has to be calibrated directly via wifi, but that's not a problem.

gmmanonymus111, 2020-03-01:
I'm glad it worked out. I forgot to mention that for some reason, GRBL (the ESP bridge) has a bit longer boot time while used over wifi, but after that you should be able to connect almost immediately.

arkypita, 2020-03-04:
> Setting the thread to fast, insane or slow, i'm seeing that the image is becoming darker than the same with serial connection.

What version of GRBL? Using M3 or M4?

gisep, 2020-03-05:
1.1f and M4, but that was due to the bad connection that sometimes slowed the laser "segments" while ON.

By the way my initial problem has been solved by opting for the Telnet connection.

I should close this discussion. Bye!

closed 2020-03-05 by gisep