arkypita/LaserGRBL #896: Raster image import for actual lasers

ballamuth, 2020-02-28:
Hi!

It's not really an issue, it's rather a feature request.

I'm trying to make a printed circuit board using a Chinese DIY laser engraver and LaserGRBL. First, a thin coat of paint is sprayed on the copper layer of the future PCB, then with the engraver the paint is removed in the zones that are subject for etching in ferric cloride.

The image for the PCB is exported from another application into bitmap, then this bitmap is imported into LaserGRBL. The problem is that the raster image import in LaserGRBL cosiders the laser to be ideal, it thinks that the laser produces an ideal dot on the surface being engraved, while the real laser diodes, even if well-focused, often (always?) produce a small rectangle, but not a dot, and the dimensions of this rectangle (at least one of them) are not negligible. For example, in my case the height of the rectangle (Y-axis dimension) is 0.25mm (only 4 lines per millimeter are enough to cover the engraved image completely), and it's too much for making a good quality mask for later PCB etching (with the existing raster to g-code translation).

In the same time the other dimension of the laser spot is very tiny, so the resolution along the X-axis is very good. And here's the idea I have in mind. First of all, I'm going to turn the laser module, so the laser spot will have significant size along X-axis, and negligible along Y-axis. Then, let's assume no traces in PCB will be closer than 0.25mm. Then comes this feature request. When translating a raster image into G-Code, let's turn the laser on one half of it's beam width later, than for an ideal laser, and let's turn it off one half of the beam width before, than for an ideal laser. I.e. laser beam should not be turned on and off directly on the edges of black and white, as for an ideal laser with a tiny circular spot, but real shape of the laser beam should be taken into account.

I think this will be useful not only for me, but for other people too.

arkypita, 2020-03-04:
I understand your request but it is too difficult to implement for the mode that LaserGRBL work actually.

tatarize, 2023-11-02:
I think lightburn added something like this a few years back. But, I think they convert their pixels directly into positions to have the little scanlines already made. So they have 3 pixel scanline, and they draw it 3 wide. But, can shorten the ends or starts of those fragments of scanlines as a whole. I believe the idea is actually because even a fine point will burn a bigger hole than the pixel. So your 2 pixel wide cut during a raster actually ended up being about 2.5-3. Really I think usually you're dealing with a general inability to have the stepper motors give that *that* fine of a control over your product. I believe the belief was that overall these extra half-pixels at the start and end of cuts where there wouldn't be overlapping made things a lot darker.

XQ6FOD, 2024-10-07:
I would also find it useful to have compensation for the actual dimensions of the laser spot. 

I would actually go one step further than Ballamuth: It would be nice to allow the user to enter the size of the actual burned spot, in terms of X and Y, and then have LaserGRBL burn all black areas staying off of all edges of these areas (in X and Y) by one half of the spot dimensions entered by the user.

I fully realize that with LaserGRBL being free software, it's totally up to its creator to decide which features to implement, and which not. But at the same time I think that many people who use their laser engravers to do precision work, such as printed circuit boards, configurable laser spot size compensation would be a very useful and welcome feature.