arkypita/LaserGRBL #899: Neje Master 20W und die LaserGRBL Software

pepponia, 2020-03-03:
Ich möchte Laser GRBL Software mit dem NEJE Master 20W benützen. Was muss ich tun? Es gibt in der Software eine Möglichkeit zu einem Arduino zu flashen. Mir ist aber nicht bekannt welche Hardware der NEJE Master verwendet. Es steht eine  Warnung vor dem flashen, dass man eventuell die Originalsoftware ruiniert. Was muss ich tun oder beachten? Bitte um Hilfe. Danke
I want to use Laser GRBL software with the NEJE Master 20W. What do I have to do? There is a way in the software to flash to an Arduino. But I don't know what hardware the NEJE Master uses. There is a warning before flashing that you may ruin the original software. What do I have to do or pay attention to? Please help. Thank you

Flying-Hammer, 2020-03-12:
Hello @pepponia 
I have just mounted the Neje 20W laser on my home-made machine a few days ago, and it works like a charm.
GRBL software is ready, you can use Laser GRBL with it
No need to flash
Change the $32 to 1 (laser enabled)
On the laser, you can forget the green wire. Red and black go to 12V.
Yellow wire is PWM. If your Arduino is a Mega, I can tell you what to change in config files
Do not hesitate, I have been helped, I can help

StuartB4, 2020-03-12:
> I have just mounted the Neje 20W laser on my home-made machine a few days ago, and it works like a charm.

I also did that with a Neje 2.5 watt on a home made machine with standard with GRBL and it worked fine, but the Neje master 20 is a machine not just the laser. He may need to upgrade the firmware because I think Neje use their own and you can't just swap it. It depends on which version he has. This may help. http://wiki.nejetool.com/doku.php?id=nejelaser_master

pepponia, 2020-03-14:
Thank You very much for helping me, but the problem is: I flashed it and I set$32 to 1. It doesnt help. It seems the machine doesnt accept the foreign software.

StuartB4, 2020-03-14:
Are you able to run LaserGRBL and connect to your machine when you click the Connect button?
If you can, goto the Grbl menu in LaserGRBL and click Grbl configuration. Then post a screenshot of the settings.

chrisdorr, 2020-04-07:
Hello @Flying-Hammer 

I just got my laser too, and i could use some help with it. I want to use it with 328p controller.

Dennisconcept, 2020-04-10:
Flying-Hammer I am new to the CNC world, I use a mega (grbl) and got the engines running.;-) . I bought a separate laser neje 20w. Can you help me with the configuration and connection of the laser? There is so much info on the grbl wiki. that I no longer see the forest through the trees.
Help me please?

StuartB4, 2020-04-10:
@Dennisconcept .
All you need to do to get that laser working is find out which pin on the mega board supplies the PWM signal and connect the laser PWM pin/wire to that.  Then add the power supply to the other 2 power pins/wires +12volts and -Ground on the laser. You don't need the temperature sensor (4th pin) on that laser to get it to work.

StuartB4, 2020-04-10:
@chrisdorr 
Which one? Uno, Nano ?

gmmanonymus111, 2020-04-10:
@Dennisconcept TTL output should be pin7 or pin8, depening on your GRBL version. [This pinout](https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega) could be helpful for Arduino mega, and [this one](https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/images/grbl-Mega-5X_Wiring.svg) if you are using a RAMPS board.

Dennisconcept, 2020-04-10:
@StuartB4 @gmmanonymus111 Thank you for the quick response. I use arduino mega (china).
the site of the laser contains pwm-freq 20khz [http://wiki.nejetool.com/doku.php?id=laser_module_308020w#installation](url)  And the rbl wiki says 1khz. do I have to adjust this. (By default, the spindle PWM frequency is 1kHz, which is the recommended PWM frequency for most current Grbl-compatible lasers system. If a different frequency is required, this may be altered by editing the cpu_map.h file.) [https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode](url) But I can't find it in the config-h folder.
thanks again for the quick response :-)

Dennisconcept, 2020-04-10:
sorry this link should work [http://wiki.nejetool.com/doku.php?id=laser_module_308020w#installation](url)

StuartB4, 2020-04-10:
@Dennisconcept 

The frequency should be fineas it is. I've changed mine in cpu_map.h before and never noticed a difference.

Are you sure you are looking in the right place for the cpu_map.h file the normal place is:
C:\Users\Administrator\Documents\Arduino\libraries\grbl  (your user name where it says Administrator.)

Flying-Hammer, 2020-04-11:
Hello hello
Yes, it is possible to use this NEJE spare part on an Arduino MEGA
I am not really able to explain how, because a friend of mine (e-Friend ?) has made the job for me.
I have just made the machine, which is the easier part of the project
Let's poke him to see if he wants to contribute 
Hello @Ingwie @bracame can you please join us and help ?

Joyeuses Pâques camarade, happy easter, github people!

Flying-Hammer, 2020-04-11:
//#define SPINDLE_OCR_REGISTER      OCR4C // d'origine
  //#define SPINDLE_COMB_BIT          COM4C1 // d'origine
  #define SPINDLE_OCR_REGISTER      OCR4A // modif bracame jcg
  #define SPINDLE_COMB_BIT          COM4A1 // modif bracame jcg

  //#define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8 // origine
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 8 // modif bracame jcg

 //#define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8 // origine
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 8 // modif bracame jcg

gmmanonymus111, 2020-04-11:
@Dennisconcept I'm alsu using a mega (and RAMPS) with the same NEJE 20W laser, and it works great. I had to tweak the PWM output though. It was on D8, and it was 12v. I've found the solution [here](https://github.com/gnea/grbl-Mega/issues/55)
You need to replace your PWM config in cpu_map.h around line 259

```// Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4C which is attached to Digital Pin 6 - Ramps Servo 2
  #define SPINDLE_PWM_MAX_VALUE     255.0 // Since it's a 8bit timer
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4A
  #define SPINDLE_COMB_BIT          COM4A1

  #define SPINDLE_TCCRA_INIT_MASK (1<<WGM41)
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41)) 
  #define SPINDLE_OCRA_REGISTER   ICR4 
  #define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 6```

Now PWM signal should be on pin6 (servo2 on RAMPS), and it should be 5V.

Dennisconcept, 2020-04-11:
@Flying-Hammer @gmmanonymus111 @StuartB4 
I try to briefly tell what I do and with what. Because maybe there are other solutions and that can help other people as well
I am an upholsterer. who knows almost nothing about electronics, I used to work with cardboard molds to draw the patterns. 

I was looking for something that I could draw or cut everything digitally and then i found the mpcnc. I made everything myself with a 3d printer and with the help of v1 engenering form. based on a mini rambo with marlin. and control everything via vp1 via wifi. the range of the table is Y2400mm and X 1800mm. but now that all this works and more and more production has to be made, it is of course crazy to have the machine drawn and then I have to cut it myself with scissors.
I went to see how the (real machines work) and that is usually with a tangential and osilating knife.
That's how I ended up with grbl and ugs. But now I want to first try with a laser to cut the fabrics. before I continue with a tangential osilating knife. Because then it becomes even more difficult (but think nice)

My hardware for this project: rp4 with manjaro linux, separate microstep drivers, 12v 4,2A power supply, nema 17 stepper motors, arduino mega 2560, neje maste 20W. Software inkscape, ugs, grbl.
The Rpi4 works great. the only problem i had was that i couldn't access the usb (can't open device "/ dev / ttyUSB0": Permission denied) with arduino ide and ugs.
this has been partially resolved. 
Open terminal: 
sudo groupadd dialout
user pasword
sudo gpasswd -a [username] dialout
sudo usermod -a -G dialout [username]
sudo chmod a+rw /dev/ttyACM0
The only problem left is that i have to re-enter this every time i restart the pi or disconnect the usb cable from the arduino from the pi. 
still quite a long story ;-)

[image]

Ingwie, 2020-04-11:
Hi,
Yes, the mod made for Flying-Hammer was switching PWM mode from 15 to 14.
With this mode (14) we use OCR4A -> Pin H3 (D6 in Arduino language).
Regards

Dennisconcept, 2020-04-11:
@Ingwie 
Sorry I do not understand it. can you explain it more clearly please?

Ingwie, 2020-04-11:
Hi Dennis,
The code modifications are described above by Flying-Hammer.

OCR4A was used to store the PWM value, by using Mode 15, this value is stored in ICR4 and we can use OCR4A (Pin H3) as output.

PWM modes are dercribed in the Mega2560 documentation [https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf](url)

-> Table 17-2 Page 145.

I hope it's clear ;-)

Flying-Hammer, 2020-04-11:
Hello @Ingwie 
Nice to see you joining this conversation !
In addition, and if it could be useful for others, I have added a little something to my setup. The NEJE's fan makes a fucking noise as soon as it is powered. Because it is not useful to power it during homing and other preliminary idle operations, I have mounted a little 12V relay on the Ramps "coolant" output in order to switch the laser ON or OFF by using a simple M8 / M9 at the beginning / end of the engraving operations (independantly of M3 / M5 ON/OFF beam)

(before M9, you can add a "dwell" to allow the laser head to cool before switching off : G4 P10000 or something
The machine suddenly becomes silent.

If you are interested, I have also added calibration and homing "buttons" to LaserGRBL

Dennisconcept, 2020-04-11:
Uhmm.
Owww mennn now i get the why, i don't understand why i can't find it.
you talk about laser-grbl and I talk about grbl-cnc.
but so far i can't find laser-grbl for linux only for microsoft windows.
how bad is this. So!

Ingwie, 2020-04-11:
Did you tried WINE ?

Dennisconcept, 2020-04-11:
no

Dennisconcept, 2020-04-11:
First I want to try to get it working with grbl-cnc. Thanks to everyone, even though I was wrong myself, I learned a lot again, Super friends! 
another question has anyone already cut artificial leather with the 20w laser?

Flying-Hammer, 2020-04-11:
Leather, no, but 20mm thick foam.
Question of power and feed.
What do U mean with "GRBL-CNC" ?
It seems you mix different concepts (or maybe am I dumb?)
GRBL is a firmware, installed in an Arduino (Uno or Mega) unable to work standalone.
To communicate with GRBL, you need a Gcode sender like cncJS (https://cnc.js.org/) Universal Gcode sender (https://winder.github.io/ugs_website/) bCNC (https://github.com/vlachoudis/bCNC)

Even LaserGRBL (http://lasergrbl.com/) is a kind of Gcode sender, including Gcode generation.

.... I hope you are aware that in your particular business, laser involves big fire risks ???

Dennisconcept, 2020-04-11:
Grbl v1.1 Laser Mode I want to try  Thank you for your concern

ciotic, 2020-04-12:
Hello everybody. I am also interested in using the neje master machine 20w with lasergrbl software.
I have GRBL 1.1f on my machine, but I think is a proprietary firmware developed by Neje. 
I connected the machine to the lasergrbl and I even used to make some contours, but when i wanted that contour to be filled, I have an entire burnt area. The laser engrave even on travel distance.
Another question would be: is it possible to use the machine end stops to home? How can I define the limits, or the start point? I know for sure that the motherboard has a 8bit chip so the pwm should be set from 0 to 255. 
Is there any way to control the power of the laser module while is engraving?( I have tried the buttons on bottom of interface to change the power but seems that nothing is happening)
Hope someone can help me with this.

Flying-Hammer, 2020-04-13:
Hello @pepponia, others
Hope you have finally succeeded in connecting your 20W spare part NEJE laser to Arduino Uno ???
If "not", I have kinda solution for U, surprisely tested last night

Open cpu_map.h
Locate this paragraph

>     // Prescaled, 8-bit Fast PWM mode.
    #define SPINDLE_TCCRA_INIT_MASK   ((1<<WGM20) | (1<<WGM21))  // Configures fast PWM mode.
    #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20)               // Disable prescaler -> 62.5kHz test 3 jcg au pif
    // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS21)               // 1/8 prescaler -> 7.8kHz (Used in v0.9) test 2 jcg au pif
    //#define SPINDLE_TCCRB_INIT_MASK   ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz test 1 jcg au pif
    // #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98kHz (J-tech laser) valeur d'origine (marche pas)

This original "define" doesn't work : #define SPINDLE_TCCRB_INIT_MASK  
The others, YES
At the moment, I use this one :     #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20) 

I don't understand what do they do and why, maybe somebody knows, I have just proceeded by "trials and errors" until finding.

I use a Protoneer CNC Shiels on which we find the PWM at "Z+" endstop pin (yes, they made a mistake in PCB wrintigs)

Test it and come give a feedback

ciotic, 2020-04-13:
@Flying-Hammer 
Can I use an Arduino Uno to acces the GRBL firmware on my neje motherboard?

Flying-Hammer, 2020-04-13:
@ciotic I don't know : I only use the NEJE laser source spare part, with the Arduino UNO of my CNC
I don't understand what you exactly want to do, but I don't think NEJE's firmware is proprietary, it would be a stupid attitude.
If you have a UNO sleeping somewhere, maybe could you try to change for it and have a full access to your machine.
Regarding endstops story, do you have any ?

ciotic, 2020-04-13:
@Flying-Hammer I have an Arduino Uno but have no idea how to use it. I bought it just to unlock the bootloader of my ender 3:))
I am pretty sure that neje is using a proprietary GRBL firmware on this machine.

Flying-Hammer, 2020-04-13:
@ciotic Unlocking a 3D printer can be a good intro to Arduinos !
 When you are in your NEJE software, do you have any way to input kind of "$$" in a console window and see what is the answer ?

ciotic, 2020-04-13:
@Flying-Hammer I can't do this in neje software, but I can do this in lasergrbl. And I receive the values from my GRBL configuration...

Flying-Hammer, 2020-04-13:
OK it seems to be a regular GRBL, good news.
The firmware (GRBL) does not seem to be proprietary. The software, yes maybe.
But I don't understand if your machine IS a NEJE or something else : travels are 150x150x200 ?
Maybe a DKxxx
Regarding the issues you are describing above (contours and so on)... are you SURE you understand and use properly LaserGRBL ??? I don't want to be unpleasant, but sometimes the problem is just between the chair and the keyboard....  A good way to know it should be done by sending the file you want to burn

ciotic, 2020-04-13:
@Flying-Hammer 
My machine is a neje master, Z should be 0 but I don't think this matters(or it is):)?
If the GRBL is not proprietary how can I make it use the physical switches to home the machine? 
It can be that I am a newbie to those things, but I like to believe that I am a smart guy and understand quite easy what is happening.
About the contours...I uploaded a PNG image, vectorize it in lasergrbl, and choose to fill it diagonal. All good, but when the laser start engraving, it engraves also on travel moves (not with full power, because I can see some differences). I will upload some photos with what I am doing in lasergrbl

ciotic, 2020-04-13:
@Flying-Hammer Here it is how i am working to this software.
What i saw now is if i am sending G90 command to move on X, i need to have maximum value of 85, in order to move the entire length of the X axis: G90X85 is the maximum travel on my machine. So i changed the configuration and put 85 on X and Y maximum travel
[neje.pdf](https://github.com/arkypita/LaserGRBL/files/4470742/neje.pdf)

Flying-Hammer, 2020-04-13:
Do not be upset, friend.
Maybe I have pointed something observing your $$'s
Please set your $32 to 1 instead of zero : this is "laser mode"
GRBL is made for milling, also. When you use a milling machine, the mill has to slowdown, stop an start again at the beginning of the next move.
In laser burning, the beam must stop and start immediately at the beginnng of each segment.
While you test this, I go on studying the message you just sent.

ciotic, 2020-04-13:
@Flying-Hammer 
I know that I need to have $32 set to 1, but I receive an error when I am trying to change that value. But the machine is working when I am using it only to make the edges...

ciotic, 2020-04-13:
Here you can see how i uploaded the 1.1f firmware made by neje. It is written that the Variable_spindle is disabled. I have the hex file of the firmware, in case someone can check it :)
[image]

Flying-Hammer, 2020-04-13:
Difficult to say what...
The main question is to know exactly what you reproach to the machine, and decide to fundamentally change the setup or not.
I you think LaserGrbl will fullfill your needs better than Neje software, you can throw away the original motherboard, put your Arduino inside and be completely independant.
In any case, it will be easy to come back to the original setup.
Do you have the skill for solderind, crimping, plugging anything ? Its not a big deal. You will have to understand also Arduino IDE and GRBL language. Nice and rich experience.
In addition, does your machine have endstops ? IMHO, endstops are important.
Voilà, voilà....

ciotic, 2020-04-13:
I would like to use lasergrbl and lightburn as well. I don't have a problem soldering crimping etc. But for using my Arduino I need a good tutorial. The endstops are something strange. There are not the same as I have on my 3d printer. They are actually small buttons(and 1 of them is directly on the mainboard) so if I am going to use my Arduino I will need also another endstop :)

arkypita, 2020-04-13:
> What I found about this error 17 is that is related to pwm settings

I can confirm what you say.

[image]

Error 17 is that is related to pwm settings. Or better: is related to variable spindle that is disabled in firmware  ([VARIABLE_SPINDLE in config.h](https://github.com/gnea/grbl/blob/5967839ab38f8e984a384db1f64e781cffb9e739/grbl/config.h#L339) of grbl). Grbl reply with this error at each try to set $32 (regardless of 0 or 1) because it is not possible to use laser mode if you have PWM disabled in firmware.

> I believe the neje made some proprietary firmware for GRBL

In the picture you have uploaded you can read the list of that changes.
Step, direction and aceleration are settings, you can change by configure proper $$ values.

The only true firmware changes is the one that disable variable spindle. Not really a proprietary firmware!

arkypita, 2020-04-13:
You can flash an official grbl to your board using grbl flash tool from LaserGRBL menu:

https://lasergrbl.com/usage/flash/

You can easly go back using upload button from neje

arkypita, 2020-04-13:
I assume that you know that in order to use variable spindle, and therefore the laser mode, your laser (laser driver) MUST NECESSARILY have the possibility of being modulated in PWM, i.e. it must have a 3-wire connection, one of which allows power modulation.

If, as I think, your laser driver does not physically have this possibility, it will not be enough for you to change a configuration data or change firmware to be able to modulate the laser power!

ciotic, 2020-04-13:
@arkypita thank you. I will try to flash it with the official version. Do you know this Neje machine? So you think it will be possible to use the physical end stops? Another problem that I have is with the axis...my Y axis should be inverted because the starting point is on top left. I don't know how to define this in lasergrbl in order to be in the good position when I add something to engrave

ciotic, 2020-04-13:
> I assume that you know that in order to use variable spindle, and therefore the laser mode, your laser (laser driver) MUST NECESSARILY have the possibility of being modulated in PWM, i.e. it must [...]

Is there any way to know for sure if my laser allows power modulation? I have no idea what should I check for this.

arkypita, 2020-04-13:
>I knew about this but I am not sure my laser can be modulated in pwm.
> LE: it seems that the 7w version support TTL, so I believe the 20w will also support

I'm confused, I don't understand why if the laser supports PWM neje should provide firmware with this function disabled.

I'm sorry but I don't own and I don't know that type of engraver, if not from the photos that are on the internet. The other owners will be able to tell you something more.

> So you think it will be possible to use the physical end stops?

The limit switches are cool, prevent the axes from going out and allow you to homing (that is very useful).

Limit switch are managed by the atmega328 chip where grbl run on. You should gain access to your controller board and see if it use a standar arduino or a proprietary board. If they use a standard arduino (uno-nano), here you can find how to connect switch:

https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

If they have a proprietary board and they did not give an header/pin for all the atmega328 pins it could be more complex but not impossible. You should reverse the design and/or wire directly to the atmega chip.

> Another problem that I have is with the axis...my Y axis should be inverted because the starting point is on top left.

You can change value of $3 to invert direction of axis
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Flying-Hammer, 2020-04-13:
@arkypita I own this 20w laser (Aliexpress sells it as a spare part for NEJE machines)
It has PWM and a 4th wire, for temperature, useless
I have succeeded in using it with a regular Arduino on a home made machine
See higher in the topic

ciotic, 2020-04-13:
@arkypita I am sure the board is a proprietary one, it doesn't look like a Arduino nano.
I will try tomorrow to flash the firmware and let you know if it worked.
For the Y axis problem I used $3=1, because this was the most appropriate, but this way I need to define my physical starting point on left bottom even if the actual end stops are on top left. I there any way to keep the end stops on left bottom but invert axis X and Y?

Flying-Hammer, 2020-04-14:
@ciotic if U really want to be involved in GRBL understanding, you should invest yourself in learning how to use Arduino IDE to check / modify / send firmware to your Arduino. This is the only way to be independant.

ciotic, 2020-04-14:
I can confirm that i flashed the original GRBL firmware, but now nothing happens(i can move the axis but without any laser beam). I can save the $32=1 but i don't have any laser beam. I returned to neje GRBL firmware and i at least i ca do some cuts now:)
I would like to understand and learn Arduino but i have no idea from where to start:)
I have a clone of arduino uno so i don't even know if i can do something with it.

StuartB4, 2020-04-14:
@ ciotic 
There's loads of help learning the Arduino.
https://www.arduino.cc/en/Tutorial/HomePage?from=Main.Tutorials
https://www.youtube.com/watch?v=TcTSXehoukU

Flying-Hammer, 2020-04-14:
@ciotic An Arduino Uno is not enough because you have to use "stepsticks" drivers for Nema 17 (Google "Pololu" or "A4988" to see if the NEJE board have any. If "yes" you can re-use them, but you can't plug them direct into the Arduino. You need a "Protoneer CNC Shield" (Google it) This shield helps you for connecting endstops, motors, pwm. It costs peanut.

ciotic, 2020-04-14:
I think is easier to buy directly a motherboard which supports GRBL :)
It anyone is interested this is how the neje master motherboard looks
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Flying-Hammer, 2020-04-14:
OK, this board is completely NEJE, you can keep it as is (or try to sell it as second hand)
You are right, you can purchase another kind of board, but I was just suggesting to enhance YOUR existing Arduino with a shield, in order to minimize operation cost.

> buy directly a motherboard which supports GRBL

ANY kind of 3D printer motherboard supports GRBL. Most of them are based on ATMEL processor like Arduinos + RAMPS shield

Schalonsus, 2020-04-22:
> @arkypita thank you. I will try to flash it with the official version. Do you know this Neje machine? So you think it will be possible to use the physical end stops? Another problem that I have is [...]

I'm with you, i have the same problems with the stock controller.
To use it with GRBL anyway you can cheat a little bit.
I wrote a small manual how to operate with 3rd party programs....
To solve the problem with zero point, just add some lines before the gcode.
Homing with stock controller is only possible with NEJE Software.
Also laser power adjustment with stock controller is only possible with NEJE Software.

To use the full potential you will need a different controller.

Here is the manual, the "Zero Point" problem is on page 4
[NEJE_Master_GCODE_Anleitung.pdf](https://github.com/arkypita/LaserGRBL/files/4515698/NEJE_Master_GCODE_Anleitung.pdf)

Alvaca100, 2020-06-13:
Sorry for asking more things but I have been trying too much things and I cannot solve any problem. I have a 5.5 laser like works really well but I bought the 20W laser to see if it was better but I cannot make it works. I am working with a hand made machine with arduino uno and CNC shield v.3.0

The laser doesn't turn and its fan do the same, no-one works 😅😅. I have measured the voltage and on supply it measures 12V and on the PWM it shows 5V when I command totally power so it seems to be working. Nevertheless, neither the laser nor its fan doesn't turn on.

I am a little bit desperate hahahaha I hope someone have an idea of what could be done... 

Thank you very much

StuartB4, 2020-06-13:
I would stick with the laser you already have if it's working. 
The 20w laser is still only 5.5 watts optical output power. The 20 watt's is input power from the power supply.  If the 20 watt one is more than 5.5 watts out, it's because it's being over driven with more current/voltage than it should have. That means it wont last long, especially if you want to do cutting with it. 

Nevertheless it should still work if you have it connected correctly. 
Do you have any clear pictures of your wiring / connections for everything.
Also are you using grbl 1.1. If so, do you have the PWM wire from the laser connected to the Z+ or Z- pin on the CNC shield.

StuartB4, 2020-06-13:
Did the laser come with a power supply, if so I would use that but also connect the ground from that to the ground on the shield, so they share a common ground.

Alvaca100, 2020-06-13:
Thank you Stuart,

I've read that it is like a 5.5W laser but as the other is also a Chinese product I am not sure if it is real or if it is slower. The idea is to find a laser which can cut at least 3mm MDF.

Maybe with these images I can explain a bit. Let's see:
[image]
[image]
[image]

I have read at Nejes wiki that I can use the 4pins as 12V+ / GRND / PWM / GRND so that is what I am doing. I am connecting the 12V wires (red and black) directly to the source power and I can measure the 12V. On the other hand I am connecting the PWM (yellow) to the left Z+ and the GRND (green) to the right Z+ as you can see on the photos.

With this configuration I can use my other laser but I can't with the Neje 20w. As I told you neither the Neje fan is working. I have also opened it a bit so I can measure in the insider driver board the voltage of the different pair of elements and they seems to work fine.

If it helps a bit I have read that I should modify the grbl config.h file. The problem is that I do not understand why should I do that if it was working with the other laser.

Also I have seen some articles that say it is necessary to use a Moffset like this one. But on other webs it is said it is not needed.
[image]

As you can see I am bit lost hahahh

[35 more replies]

closed 2020-12-16 by arkypita