arkypita/LaserGRBL #901: vg-l7 and laserGRBL

domcrat, 2020-03-04:
Good afternoon. The problem lies in the incorrect operation of the engraver in the laserGRBl program. When engraving a photo, I can 't choose M4, I only always have M3. The command $32=1, it accepts if it is registered. But M4 does not appear. maybe you know what's going on . I attach a photo of the module. SWLD 59 V3.5 processor. It connects via a 3 pin wire, but there is a free 2 pin hole on the Board . maybe you know what it is for and you need to connect it ? I must say that the laser power is currently regulated.
[image]
[image]

arkypita, 2020-03-04:
What version of grbl do you have? it should tell you here when you start LaserGRBL and click connect.
[image]

domcrat, 2020-03-04:
My version 1.1f is $30=255, when I turn on the program, I always have $32=0, I constantly change it to $32=1, the program writes OK, but M4 does not appear in the list that I did not do.

domcrat, 2020-03-04:
check the box for PWM support. What is the probability that this is due to the 2 pin connector? can this be? the instructions about this do not say anything, maybe you know based on this processor

arkypita, 2020-03-04:
To see M4 you need GRBL version >= 1.1 with laser mode enabled ($32=1)
No other information is known by LaserGRBL about your hardware.

> when I turn on the program, I always have $32=0

I have a vague recollection that someone has already reported a similar problem, but I can't find the post. If I remember correctly he had solved by rewriting the GRBL firmware on his controller board.

Just to be sure, do this test and let me know.

1) connect lasergrbl to the board and you put in idle state
[image]

2) Open the grbl configuration window, be sure that it show "34 config readed succesfully" when it open.
[image]

- Tell me the value of $32.
- If the value is zero, set it to 1 and push "Write" button.
- Push "Refresh" button and check if is correctly set to 1 or not.
- If is correctly set to 1 please close the grbl config window, disconnect, and close LaserGRBL.
- Re-open LaserGRBL and connect (idle)
- Open grbl configuration (34 config readed succesfully) and check if the $32 is 1 again or not.

Let me know.

domcrat, 2020-03-05:
Good day arkypita. I tried to change the version on your recommendation. 1.1f to 1.1h
[image]
[image]
[image]

. as a result, after driving in 32=1, and manipulations that you recommended, the setting was saved and M4 appeared. Thank you for your help.
[image]

arkypita, 2020-03-05:
You are welcome

closed 2020-03-05 by arkypita