arkypita/LaserGRBL #903: Limit Switches 3018 Pro

algot1901, 2020-03-05:
How do I control the machine from hitting the of travel? Also when it is at the end of travel the motor sounds like it is going to destruct and my only recourse is to shut the machine off.

gisep, 2020-03-05:
With limit switches

algot1901, 2020-03-05:
Is there schematics to show how to use the limit switches I know in Mach 3 when it hit the switch it would reverse a bit also how does one set the home position and also the x and y as I have no idea how this grbl works

gmmanonymus111, 2020-03-05:
[This](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches) might help you getting started
Home position is set by homing your machine (in LaserGRBL, a house icon, if homing is enabled)

algot1901, 2020-03-05:
Another question, the GRBL $$ Configuration screen how and where do I add the homing and limits to the program I am guess that the limits would have to be written in for each drawing.

gmmanonymus111, 2020-03-05:
GRBL by default does the homing in the following order: first, the Z axis, then X and Y simultaneously.
If you don't have a Z axis limit switch, you need to recompile GRBL from source to disable it :/ (or you can flash the one with LaserGRBL, I don't know if it is disabled there by default)
Note: you need to enable hard limits in GRBL config (I don't know the $code for it, but with LaserGRBL, you can check every config option, once connected)
For soft limits (without limit switches and homing), you need to set your machine's dimensions (X/Y max travel)

closed 2020-12-16 by arkypita