arkypita/LaserGRBL #904: Connecting 7 Watt laser to Arduino Uno with CNC Shield

JCowper, 2020-03-05:
Hello Everyone,

So I am brand new to this kind of stuff and I am currently trying to fix my laser CNC. The original board was destroyed in an accident and I thought it would be easy just to replace the board with an Ardiuno Uno with a CNC Shield v3.00. I have I am pretty sure I have wired up the stepper motors properly, but I am realizing there isn't a 3 wire port to plug in the laser. It is a 7 watt laser and the wire off of it is red, white with yellow stripe, and black. I have 12 volt 5 amp power supply from the last board that can plug into the the Ardiuno Uno directly if needed.

At this point, I am not sure what to do. I have tried to understand the forums but it all sounds like garble to me. If someone can explain this or draw me a picture that explains this in simple terms that would be greatly appreciated.

Thanks,
JC

JCowper, 2020-03-05:
Those that may be interested, this is what I have so far.
[image]
[image]

gmmanonymus111, 2020-03-05:
You will need that power supply for the laser, as arduino can output a maximum of 5 volts.
Laser wires should be the following (hard to tell without a model number):
Red  - 12v - from external supply
White/yellow - ttl - to arduino pin 11 or z limit on cncshield 3
Black - gnd - arduino or external supply gnd (I'm not sure which one is needed for ttl)

arkypita, 2020-03-05:
https://github.com/arkypita/LaserGRBL/wiki/Laser-Wiring

JCowper, 2020-03-05:
Hmmm ok so my power supply shouldn't be plugged into the Arduino Uno? Or do I need a separate power supply just for the laser now?

Current power supply:

[image]

gmmanonymus111, 2020-03-05:
The uno should be fine from a 5V phone charger
Power supply can go to the laser and the shield

JCowper, 2020-03-05:
Only input options available:
[image]

gmmanonymus111, 2020-03-05:
Phone charger adapter <- arduino data cable -> arduino data input
12v power supply -> shield input (that blue plastic thingy with 2 screws)

JCowper, 2020-03-05:
So if I plug the Arduino into my laptop through the USB port, that should be enough to power it?

Also, if I cut the tip off the power adapter and wire it into the blue port, where do I wire the laser into?

gmmanonymus111, 2020-03-05:
It should be wired to the power supply, so I'd say it should go to the blue port too, at least that's how I've done it

StuartB4, 2020-03-06:
Yes you can just power the Arduino from the USB. It does'nt need any more than that.
The 12 volt power supply you have in your hand you can use just for the laser power Red and Black.

Get a seperate 12volt / 5 Amp power supply for the CNC Shield. That powers the stepper drivers and motors. 12v is enough, but you can use one up to 36v because you have the purple DRV8825 drivers on the shield. Connect that to the Blue power connector on the shield.

Connect the Black -ve leads from both power supplies together. That way everything shares a common ground. I know some people have had issues with the laser not powering up properly until they did that.

Then connect the White wire from the laser to either the Z+ or Z- pin on the shield, it does'nt matter which one, they are both connected together on the shield's circuit. That gets the PWM signal needed to vary the power of the laser from Pin 11 on the V3 shield that you have.

JCowper, 2020-03-15:
Hello everyone,

So I had some time today to try and wire this laser again. Unfortunately it didn't end very well, as I blew all the capacitors on the CNC shield. So, where did I go wrong? In the meantime I am ordering a new CNC Shield.

Currently, it is wired so that one 12V 2A wall adaptor is is screwed right to the board. I have another wall adaptor that I have spliced directly with the the red and black from the laser, and the yellow plugged into Z- on the white side.
[image]

gmmanonymus111, 2020-03-15:
Hello,It's sa to hear... Are you sure you wired the 12v adaptor the right way? It seems the capacitors got reverse polarity, which caused them to explode. Reverse polarity might damaged your laser too, if it doesn't have protection against that kind of damage (most lasers fortunately have one).

StuartB4, 2020-03-16:
I agree with gmmanonymus111, maybe you have the power leads the wrong way on the shield. 
Also, the stepper drivers are definately the wrong way round.

JCowper, 2020-03-19:
Thank you everyone so far. I just had a new CNC Shield come in and I have rewired it the right way. Next problem. Everything is turned on, but the Ardrino is not connecting to the GRBL program. Any hints as to what I am doing wrong this time? LoL

This is all a learning process for me.

Thank you in advance again for your help.

StuartB4, 2020-03-19:
You could have damaged the Arduino when the capacitors blew. 
Also, did you put the stepper drivers in the correct way this time?

JCowper, 2020-03-19:
Steppers are in the right way, and I have just finished re-flashing the new Uno through the GRBL software. Now it is connecting, but it isn't responding to the G-Code. I tried to get it to move around and it isn't moving in any direction.

StuartB4, 2020-03-19:
Does it try to move at all, any noise from the motors?

JCowper, 2020-03-19:
Motors make a sound but don't move. It is like it is trying with a thumping sound, but no movement at all.

Also, the laser is on, it is burning, but I can't seem to control it either. I had read above that the yellow has to be plugged into z- but is it supposed to be plugged into the white or black side.

StuartB4, 2020-03-20:
The white side. all the pins on the black side are ground.
You might need to join both the ground (-'s) on both power supplies together to make sure the laser pwm signal gets a proper ground.

Have you set the stepper motor current on the driver boards?
You can see how to do it here.https://www.youtube.com/watch?v=89BHS9hfSUk
 I set mine to about 300 millivolts.
Although you are setting the current you actually do it with the 2 volt DC voltage range on a multimeter.

closed 2020-12-16 by arkypita