arkypita/LaserGRBL #905: I have a issue with the speed and burn of my lasaer.

Xander1983, 2020-03-09:
I am having an issue with the speed and burning of my laser. When I got it the stepper board wasn't working right so I got a new one. In the process I had downloaded other things to flash and update my arduino nano v2.5 board. After I got the new stepper motor it and put it in my board still wasnt acting right so I updated the firmware again and it worked perfectly. Just was burning to big of lines so I tried to fix the laser and in the process I went into lasergrbl and edited some setting that I saw a guy using online. Now my stepper motors are moving really fast and my laser just wants to create dots. I have no idea how to get it to slow back down or burn a straight line.
[image]
[image]
[image]
[image]

arkypita, 2020-03-09:
$100 $101 set how many motor step is needed to move the head by 1mm
They look too high for me, usually they are in the order of 40-200 step/mm

The exact value depends by your hardware, Use stuart excel file to fine trim exact values
https://github.com/arkypita/LaserGRBL/issues/589

StuartB4, 2020-03-09:
Also, I noticed you have $30, 31 and 32 in your grbl settings,  change $32 to 1 to enable Laser mode.
You can leave $30 set to 255.

closed 2020-12-16 by arkypita