arkypita/LaserGRBL #907: SVG files does not appear to be imported/created correctly

Ziggy2013, 2020-03-10:
SVG file does not appear to import correctly.

The file is for a solder paste stencil. The svg file displays properly in Inkscape but does not display properly in LaserGRBL.  A few of the solder pads are not in the correct place and appear to be mirrored around an arbitary XY axis. 

SVG file attached - renamed to *.txt as svg files are not accepted

[PC_wifi_19_brd_mod.txt](https://github.com/arkypita/LaserGRBL/files/4309814/PC_wifi_19_brd_mod.txt)

Ziggy2013, 2020-03-10:
LaserGRBL Version is 3.1.2

arkypita, 2020-03-10:
See if this can help you: https://github.com/arkypita/LaserGRBL/issues/872

gmmanonymus111, 2020-03-11:
I've played around with Inkscape (v0.92.4) and it seems rotating an object makes its position messed up.
Before rotating:
[image]
After rotating:
[image]

With v0.91, it seems to be working correctly.

Ziggy2013, 2020-03-11:
Thanks gmmanonymus111 for following up.
Resizing an object also causes Inkscape v0.92.4 svg files not to import correctly into LaserGRBL.
I have resorted to editing the svg file XML statements directly so the LaserGRBL import works.
At this point I haven't identified what Inkscape generated XML construct causes the LaserGRBL import issue.

arkypita, 2020-03-11:
very nice color scheme

Ziggy2013, 2020-03-12:
I believe the root cause of LaserGRBL not importing the Inkscape svg files correctly is that Inkscape v0.92.4 uses an XML syntax which combines both translation and rotation.

if the XML syntax is separated into a translation and a rotation step (eg like this)

 transform="translate(22.387291,-39.921669) rotate(-270.000000)"

LaserGRBL imports correctly.

However in Inkscape v0.92.4 the translation and rotation are combined (eg like this)

     transform="rotate(90, 31.15448, -8.767189)"
  
LaserGRBL does not appear to handle this syntax correctly.

By directly editing the Inkscape v0.92.4 svg to separate the translate and rotate steps LaserGRBL will import correctly.

Happy for anyone else to confirm I'm right or prove me wrong. For example, I suspect that Inkscape 0.91 does not use the combined XML construct but I haven't checked

Ziggy2013, 2020-03-12:
I have since realised that an example like

transform="rotate(90, 31.15448, -8.767189)"

is not combined rotation and translation. It is specifying rotation about a point. A good reference is here

http://tutorials.jenkov.com/svg/svg-transformation.html#translate

But the issue still is that LaserGRBL does not appear to import this XML syntax correctly.

arkypita, 2020-04-08:
Thanks @Ziggy2013 for your analysis about rotation
I will try to see if I can correct this issue.

arkypita, 2020-04-08:
I close, let's continue this topic in the general issue on SVG support #451

closed 2020-04-08 by arkypita