arkypita/LaserGRBL #908: Laser not working

ramyasreeadari, 2020-03-10:
I have bought CNC 3018 pro machine . I have mounted the laser and gave all the connections accordingly, and installed all the software and drivers in my laptop .now I started opening the software checked all the setting and gave the test size than the laser spot shows the test size on the workpiece,later on by clicking on the start button for laser engraving the X And Y axis adjust themselves to start the work ,the program is running but the laser is not burning as soon as I click the start button

taz101, 2020-03-23:
See if you can increase power to 1000. 255 did not work for my laser.

chasehensley, 2020-03-23:
> See if you can increase power to 1000. 255 did not work for my laser.

Hey different person but! I am also having this issue and mine works when I set it to 1000 but not any lower why is this?

StuartB4, 2020-03-23:
@ramyasreeadari.
I think you have to enable Laser Mode, $32 on that machine to use the laser, then disable it to use the spindle motor.

mjscholp, 2020-04-09:
> @ramyasreeadari.
> I think you have to enable Laser Mode, $32 on that machine to use the laser, then disable it to use the spindle motor.

how do you do that? Right now it is set to 1, i thought that was enabled?I just purchased this and am new to all of this. I am unsure if anything is enabled or disabled.

StuartB4, 2020-04-10:
Yes, set to 1 is enabled. 
Also $30 should be 255.

mjscholp, 2020-04-10:
> Yes, set to 1 is enabled.
> Also $30 should be 255.

Yep, that's what I have and still nothing. Thank you

StuartB4, 2020-04-10:
@ramyasreeadari 
@mjscholp
Can you post some clear pics of your controller board and laser showing the connections.

mjscholp, 2020-04-10:
[image]
[image]
[image]
[image]

StuartB4, 2020-04-10:
@mjscholp 
Sorry I cnat read any of the writing on the underside of the board, too blurred. Also, theres a white wire coming from the laser which is usually the PWM wire, but I can't see where it goes to on the board, does it change colour somewhere?

mjscholp, 2020-04-10:
[image]
[image]
[image]
[image]

I dont think they change colors. They called out the pin# for connection, not saying that something couldn't be wrong but it is installed and connected per the instructions. Ill keep looking and see what i can find. thank you for taking the time.

mjscholp, 2020-04-10:
> @mjscholp
> Sorry I cant read any of the writing on the underside of the board, too blurred. Also, theres a white wire coming from the laser which is usually the PWM wire, but I can't see where it goes to on the board, does it change colour somewhere?

I found the issue, of course user error. there was 1 connector not connected! thank you for taking the time

StuartB4, 2020-04-10:
Ok mjscholp. 
Glad you got it sorted. :+1:

closed 2020-12-16 by arkypita