arkypita/LaserGRBL #912: Permanently store work offset

Flying-Hammer, 2020-03-12:
Hello
I don't know if this subject has already been evoked but...
IMHO something is missing in LaserGRBL, ability to permanently store the last work offset used.
The "globe" button allows to make a G92, nice but if U switch OFF, the home is lost
For sure U can write this offset on a post-it note and make a G0 Xxxx Yyyy Zzzz move, then G92 again, but it's not very handy.
I have circumvented this by adding a button "G92 X0 Y0 Z0" then "G28.1" which stores in GRBL the position determined by G92
And another button "G28" then "G92 X0 Y0 Z0" which makes the machine come back to work offset after initial homing
Maybe could U put something like this in your Todo list ?
Regards, and congrats for this nice job
JCG from France

Flying-Hammer, 2020-03-20:
Nobody interested ? @arkypita  ?

StuartB4, 2020-03-20:
How about adding something in here.

[image]

Flying-Hammer, 2020-03-27:
@StuartB4 @arkypita 
Adding "something in here" won't solve the issue I am mentioning
Most NC machines have endstops used to make homing. Homing allows you to retrieve your part origin, even if you unplug the PS. Saving part origin can't be done automatically by any kind of software, only by the brave operator !

StuartB4, 2020-03-27:
Oh ok.

regpye, 2021-03-21:
It would be a very handy feature for me also.