arkypita/LaserGRBL #913: Language not working

Blindbard666, 2020-03-12:
Hi!
I compiled the project right after installation. Everything works fine except for the language menu: i pick a language, the application restart but it's always in English.
How can i try to solve this?
Thanx

Blindbard666, 2020-03-12:
i meant, "right after downloading it"

arkypita, 2020-03-12:
Hi @Blindbard666 
I am the developer of LaserGRBL and I have exactly the same issue as you.

On two of three PC (my office PC and laptop) when I compile all is fine, but if I compile from my home PC the language choice does not work. I cannot figure out why.

I think that is a problem of the visual studio compiler, because the problem seem to be in the compiled .exe (.exe compiled on office PC work well in all the system, .exe compiled with home PC does not work in all the system).,

Blindbard666, 2020-03-12:
thanx! I'll try on another pc then... :)

Blindbard666, 2020-03-12:
nope... i tried on 3 different pc but the issue is always there. 
All the pc have win10 and the latest visual studio release. Maybe it could be because of the o.s. or the vs version?

arkypita, 2020-03-12:
I use W10 too (both on home and work PC) My VS

Blindbard666, 2020-03-13:
same problem with win7... so there must be something in the visual studio version... it's very weird

Blindbard666, 2020-03-13:
win10 + visualstudio 2015, and it works! :)

arkypita, 2020-03-13:
Fine!

However I never realize why we have this issue.
The line of code that set the language for the UI is the line 22 of file program.cs

```
System.Globalization.CultureInfo ci = (System.Globalization.CultureInfo)Settings.GetObject("User Language", null);
if (ci != null)
	System.Threading.Thread.CurrentThread.CurrentUICulture = ci;
```

The value is correctly stored and retrieved from settings via the `GetObject` call, and the assignment 
`CurrentThread.CurrentUICulture = ci` seem to be ok, but uneffective. If you have any idea...

Blindbard666, 2020-03-14:
i maybe found out where's the problem!
win 10 and visual studio 2019: i changed the framework to .net 4.5 and it worked!

daje! :)

arkypita, 2020-03-14:
I use to compile for framework 3.5 because it is the last that is supported under Windows XP

Blindbard666, 2020-03-16:
scusa se scrivo in italiano e se vado off-topic, ma non so come contattarti privatamente: lavoro per una ditta di Lecce che si occupa di prodotti per la grafica e la stampa. Il mio titolare mi ha chiesto di contattarti per proporti un lavoro (se fattibile), una modifica a questo software che faccia sì che, caricata l'immagine, la prepari per il taglio con polistirolo come fa potrace, ovvero con le linee di unione. Allego esempio di come è ora e come dovrebbe essere (nell'esempio è caricato un gcode creato con potrace)
Si può fare? e se sì, a che prezzo? Per contattarci privatamente puoi usare la mail info [at] mybay.it
Grazie comunque!
[image]

[image]

closed 2020-03-24 by arkypita