arkypita/LaserGRBL #914: Software wont open

mommameg970, 2020-03-13:
I am new to this and i had my laser working just fine last night.
I was changing some settings and something in the global configuration got changed and i got a pop up saying if i quit it will close if i hit continue it will attempt to continue, well i hit quit and the program shit down and i have not been able to get it to start back up.  Is there a way to completely reset the laser or anything?
I have troubleshot with Microsoft and installed and uninstalled as many times as i can and i jeep having the same problem except it wont even open at all after i reinstall.
Any ideas, i am in need of major help.
Thank you

arkypita, 2020-03-13:
did you have any message talking about LaserGRBL in windows event log? (press start menu and type event, then open event viewer

[image]

I had something similar just today, solved simply rebooting windows and apply last windows update. I thin i had an update in progress that broke .net framework, something microsoft related

arkypita, 2020-03-13:
I had messages like this in my event log

[image]

arkypita, 2020-03-13:
If you need to remove LaserGRBL you can delete the folder: 
- C:\Users\Your User Name\AppData\Roaming\LaserGRBL
- C:\Program Files (x86)\LaserGRBL

mommameg970, 2020-03-13:
[image]

mommameg970, 2020-03-13:
[image]

mommameg970, 2020-03-13:
i deleted it have not reinstalled or anything
I did look at the event log as i showed you above, but i have no idea what to do now... like i said it said something about the global configuration and then wont open at all...

mommameg970, 2020-03-13:
Ok, so i got it to open in LaserWeb but i am getting a warning saying Gcode is out of machine bounds, can lead to running work hault... i assume this is why i cannot get it to open in LASERGRBL, anyone have ANY idea how to fix this?

closed 2020-03-24 by arkypita