arkypita/LaserGRBL #916: Alarm/Stopping after first cycle

MaLe80, 2020-03-14:
Hello and first of all: thank you for the great software!

I am completely new and have the problem, that the laser or software is stopping after the first cycle. When I want to cut wood I have always to make one cycle and repeat it manually. I use an Ortur Laser Master 15W. Everything is fine but this happens all the time. I use vectors for this cutting.

Any suggestions? I am not a programmer. So easy language would be appreciated. :) 

Best wishes,
Matthias

MaLe80, 2020-03-14:
Additional information: I use LaserGRBL v3.1.2
I updated the Ortur with the new firmware.

StuartB4, 2020-03-14:
Even if you set the number of passes?
[image]

MaLe80, 2020-03-14:
It is the only way I know for cycles (passes).
Is there another way?

StuartB4, 2020-03-14:
Not as far as I know. Have you tried changing this setting to a slower one.

[image]

MaLe80, 2020-03-15:
Hello,

yes, I already reduced it to "Fast". Maybe I have to reduce it even more.
But the strange thing is, that it mostly happens when one cycle is finished.
I also tried all Streaming Mode settings. Did not help.

MaLe80, 2020-03-15:
Even with "Slow" it does not work. Here is the message I get. 
On the left you can see in blue: "MSG:Reset to continue". I guess this is important but I don't know what to do.
[image]

StuartB4, 2020-03-15:
You should be able to click the Lightning icon next to the padlock to reset.
Do you have anything in the settings that are adding extra gcode commands to your file.
If you do make a note of them so you know what they are, then remove them and try again.
You can always replace them if you need to.

[image]

arkypita, 2020-03-15:
> The cause is: StopResponding
> 
> yes, I already reduced it to "Fast". Maybe I have to reduce it even more.
> But the strange thing is, that it mostly happens when one cycle is finished.
> I also tried all Streaming Mode settings. Did not help.

Yes, it is very strange.

Stop responding is signaled by LaserGRBL when grbl (the firmware that is running on controller board) does not reply to status query for more than an amount of seconds.

What version of grbl do you have on your board?

MaLe80, 2020-03-15:
Hm, I don`t know. How can I find out? Haven't found any "information"-button for this. I only see the "Flash GRBL Firmware" under "Tools".

StuartB4, 2020-03-15:
Should tell you here.

[image]

MaLe80, 2020-03-16:
Hello, 

I also have Grbl 1.1f. Shall I change this?
But I am not sure which Arduino board I have. 
I am a little bit Arduino experienced. But on the machine, it does not loke like any Arduino I know. 
It is small and long so maybe a "nano". But not sure. 

[image]

StuartB4, 2020-03-16:
No need to change grbl, 1.1f is fine. Yes it may be a Nano if it's plugged in to the main control board.
It may be Ortur's own board. Do you have a picture?

Also, can you upload a screenshot of the GRBL settings.
Just start LaserGRBL and click Connect.
Goto Grbl menu and click Grbl configuration.
When the list pops up just stretch it a bit to show all the $ settings then get a screenshot
and post it here.

MaLe80, 2020-03-16:
Hey,

thank you for your great help! It is nearly impossible to take a picture of the board without disassembling the Ortur Laser Master. Is it necessary if Grbl 1.1f is fine?

I have made the screenshot. :) 

[image]

MaLe80, 2020-03-16:
I think I have solved it. I found the menu "multiple passes". I have seen that there is a Z axis.
I deleted everything and wrote "G90" in this. Now it makes multiple passes.

[image]

closed 2020-12-16 by arkypita