arkypita/LaserGRBL #923: continuous jog movement

tittopower, 2020-03-18:
I have an issue if this function is setted to work. In normal condition (without continuous jog) after an homing setted as standard grbl $23=0 to the max run of the Xand Y axis I can move the axis to any point of the workspace. If continuous jog is enabled after an homing cycle the machine no longer moves. $130 an 131 are correcly setted and work well in step mode.
Thanks.

arkypita, 2020-03-24:
do you have grbl v1.1 or 0.9?

tittopower, 2020-03-25:
Hi ! I am usin 1.1g version of grbl. Thanks.

tittopower, 2020-03-25:
Here is my configuration

$0=10 (Step pulse time)
$1=255 (Step idle delay)
$2=0 (Step pulse invert)
$3=6 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=2 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=1 (Soft limits enable)
$21=0 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=3000.000 (Homing search seek rate)
$26=25 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=255 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=162.648 (X-axis travel resolution)
$101=162.468 (Y-axis travel resolution)
$102=3200.000 (Z-axis travel resolution)
$110=5000.000 (X-axis maximum rate)
$111=5000.000 (Y-axis maximum rate)
$112=50.000 (Z-axis maximum rate)
$120=500.000 (X-axis acceleration)
$121=500.000 (Y-axis acceleration)
$122=10.000 (Z-axis acceleration)
$130=288.000 (X-axis maximum travel)
$131=333.000 (Y-axis maximum travel)
$132=50.000 (Z-axis maximum travel)

gmmanonymus111, 2020-03-25:
GRBL normally sets your work coordinates in negative value, and LaserGRBL is trying to move from `0` to `maxTravel`. Since it is negative, it can go towards `0`, but won't go to `-maxTravel` way

tittopower, 2020-03-26:
When I start the machine and connect it I made homing first of all (right-far). The coordinates I read on lasergrbl (high right of the window) are X1 Y1 Z0 ($27=1,000)  on the upper line and X+My space Y+Myspace in the lower line. Is OK ?
Then if I moove in step mode (for example 20) I see the program send G91X20.0 Y20.0 Fxxxx and the machine moove to me regularly (left-near), the values of first line go in negative and values of second line (positives) decrease. Is OK ?
In the same conditions described I enable "continuous jog"
If now I push the arrow to moove the machine to me in continuous jog I see the command sent is  G53X0.0 Y0.0 Fxxxx, is correct ? 
The machine  do only a  movement from X1 Y1 to X0 Y0 (opposite movenete respect the arrow) and go to press the end stop switches  and stops there.
 
Where is my error ?
Thanks

jletras1985, 2020-04-30:
Please help me, my laser engraver Totem 2500mw did not engrave the real dimensions off the object. Exemple: 6x9 cm reduce +- 50%. Can you give me the correct configuration? Thanks!!

StuartB4, 2020-05-01:
First start LaserGRBL and click Connect. Then click the Grbl menu and click Grbl configuration. Look down the list that pops up for $100 and $101. Make a note of the numbers in the value boxes for both of them. You need to know them for the next step.

Then try this to calibrate your stepper motors steps per mm.
https://jscalc.io/calc/tHjfIiKMKjDF42Hg?fbclid=IwAR219p6S0qWzVAsY8aLC8nyQaXaMpfa-mSTNCIXA7TAMPa3VzrGoCDcimsI

jletras1985, 2020-05-01:
Thanks, WHen you Said “jog an axis...” -input ... how can i do that? I’ve buy the machine yesterday ....Sorry! Help me!!!

StuartB4, 2020-05-01:
[image]

closed 2020-12-16 by arkypita