arkypita/LaserGRBL #924: Alarm on connecting after upgrading firmware to 1.1h

oliverhbailey, 2020-03-18:
Hello,
I updated the firmware to 1.1h several months ago. The version of LaserGRBL at the time worked fine with the 1.1h firmware. I went to use the machine today and LaserGRBL updated to the latest stable version. When I went to connect, it triggered an immediate alarm. I don't see any open issues with this type of problem. It's a Windows 7 Pro machine. I uninstalled the serial drivers and reinstalled them. It connects, it just triggers an alarm, I suspect due to the firmware version.

I connected with Putty and fed the machine some manual movement commands. They worked, but I did not go down the list of all commands.

Any suggestions would be greatly appreciated.

Thank you,
Oliver

arkypita, 2020-03-19:
> When I went to connect, it triggered an immediate alarm

What kind of alarm

oliverhbailey, 2020-03-19:
How do I tell? I read through the documentation and don't see anything on
how to read the alarm

StuartB4, 2020-03-19:
Can you get a screenshot of what it says and post it here?

oliverhbailey, 2020-03-19:
[image: image.png]

Here is the screen shot of an attempted connection.

oliverhbailey, 2020-03-19:
For some reason the upgrade starts locked. I have been able to connect, set
home, image size, etc, but once the file starts to send, nothing happens on
the unit, There is no data being sent from LaserGRBL.

oliverhbailey, 2020-03-19:
I spoke too soon, once I disconnected and reconnected, the Alarm pops up
again.

arkypita, 2020-03-19:
Screenshot is not visible, you cannot post image by reply to the email, it work only if you post it into github site here https://github.com/arkypita/LaserGRBL/issues/924

oliverhbailey, 2020-03-19:
I'm going to uninstall LaserGRBL and use another product. I've been a
hardware and OS developer for 46 years, and the problem is with the serial
driver that accompanies LaserGRBL. I use this on a Windows 7 laptop, and it
has worked flawlessly for several years, through all the updates. But now
it send Windos 7 into a tailspin attempting to update the serial driver and
then removing the driver due to a failed update.

As much as I like LaserGRBL, attempting to support driver install and
uninstall is a bad idea. These drivers are questionable anyway because FTDI
sabotaged the driver a long time ago. The other problem is that w
supporting too many advanced features from an international version of
Windows. It just causes problems when going between languages, and always
has.

I use several products for laser engraving and may try to go back to an
earlier version on github to get things working again. I hope you get these
problems worked out because I think these issues are going to propagate
quickly if you don't. If I had the time to fix ths, I might add some help
but in all honesty, I just don't have the time.
And this is one of the big problems with open source software, it gets
developed to a point where it should be forked to baseline a working
version and ends up being over developed until it doesn't work and can't be
debugged.

Good luck with this project.

Regards,
Oliver Bailey

StuartB4, 2020-03-19:
> These drivers are questionable anyway because FTDI sabotaged the driver a long time ago.

Thats why I use a genuine Arduino UNO on my machine. It has the proper  Atmega16U2 programmed as a USB-to-serial converter. Never had a problem with it, unllike the clones with the CHR340 etc.

oliverhbailey, 2020-03-19:
Really,
Let me tell you Stewart, this worked flawlessly until yesterday. Drivers
and all. Did I purchase an Arduino clone intentionally, absolutely not.
There are also a number of clone manufactures that legitimately make
Arduino compatible clones, Inland for example, which is a division of
Microcenter. The problem isn't with the drivers. The CH40 driver is
actually being installed by Windows Update, which only downloads Windows
certified drivers. The problem isn't with the Arduino clone, and this one
came with the Eleksmaker Laser Engraver, so it was not a choice. I have t
partially working, and can tell you the problem is 100% in your software,
it's not hardware related at all. If you received a copy of my first email,
I pointed out that I connected to the board with putty and manually ran the
machine from the command line. I can fix this issue in your software and
will for my use. It's not a difficult fix, it takes about 10 minutes. I had
to go back to an earlier version of lasergrbl that I knew worked to find
the problem in the current version.

As far at FTDI goes, they thought they were protecting themselves when the
CH340 came out because the chip is a direct clone of the FTDI232R. What
they didn't realize then, is that they hurt their customers who had
unsuspectedly purchased FTDI chips not made by FTDI. Yes, the original
chips were packaged to look like FTDI chips.

There is nothing wrong with the Arduino NANO clone, ist has, and is working
flawlessly as it has for over the past two years. There is nothing wrong
with the driver. There was an outstanding Windows update that I needed to
install and once that happened the driver worked as it should. There is a
problem with how lasergrbl initializes and how the settings are stored. I
need say no more because I've seen this problem before and know the cause.
You will will eventually find the cause if you pay attention.

I use several other products that are purchased products and they have no
problems with the driver or the Nano board. Good luck and thanks for your
input.

Regards,
Oliver Bailey

StuartB4, 2020-03-19:
It's not actually my software. The developer is Arkypita who replied to you earlier. But I've been using LaserGRBL since it was first released and try to help others if I can, especially people new to the software. I do have a couple of clone/dubious looking Nano boards and they work fine with LaserGRBL with the CH340 driver, I don't like them though because they have the weak mini USB connector on them. If I buy a new one I usually put extra solder on the legs of the socket shield. I've seen a lot of people break them off the board. They are cheap enough to just buy a replacement and hardly worth repairing, it's just inconvenient when it happens. 

I use Windows 7 Pro on my machine but learnt long ago not to accept all the updates MS throw at us. My latest update from MS was when I reinstalled Windows about 5 years ago, but then computer I use on my engraver does'nt get connected to the internet so I dont need a lot of the so called security updates. I understand everyone's computers and engraving machines are slightly different but LaserGRBL works fine for me (any version) with either my Arduino UNO or the clone Nano's.

oliverhbailey, 2020-03-19:
I'm not an Arduino person, I like the PSoC, and wrote several books about
that controller. I was designing big CNC machine tools back in the early
1970's before changing careers to computer hardware and embedded software.
Most of what I do these days uses embedded Linux, and I've been
developing custom Linux systems since 1995. I was inoled with the design of
Windows NT .0, 3.5, 2000, XP, VISTA, and Windows 7.0. Mostly deice drier
interfaces and kernel interrupt schemes.

Windows is officially going away in October 2025. So where is Windows
going? If you activate WSL in features on Windows 10, you'll be able to
see, Windows Subsystem for Linux. I do real time systems, so lasergrbl is
not that high speed for me. But its suitable for many people. The Arduino
is 25 years old, so its not new technology, its very old technology.

I'.m semi-retired, heading quickly for 70, so I do this to relax and have a
good time. LaserGRBL isgetting oer complicated due to the addition of too
many features. Nice to meet you Stewart, thank you for your input and
follow up. Hae a great day.

Oliver Bailey

arkypita, 2020-03-19:
Dear Oliver

LaserGRBL is used by around 30,000 people worldwide (146 countries). It is used on Windows 10, 7, 8, XP, Vista, Windows Server operating systems, various 32 and 64 bit versions, and different service packs. The latest version 3.1.2 is used by 16'428, more than 50% of the users of LaserGRBL and there have been no major changes compared to the previous version that can justify communication problems, or other problems.

I am convinced that if you had any problems with the updates, it is due to an external cause, probably not attributable to LaserGRBL itself.

I can help you understand what this problem is, and solve it, but as a developer as you are you should understand well that stating "I have an alarm" and not adding further details on what this alarm is, what is the alarm source, where it is shown, etc does not allow me to help you.

Keep in mind that LaserGRBL itself has no alarms, at the limit it reduces to reporting those that are sent to it by the machine (i.e. from the grbl firmware).

So if you can, give us more information about this problem.

I'm not sure I understand the FTDI speech

I can't understand the tone of your speech. If it's controversial, forget it, because I'm not in the mood. Conversely, constructive criticisms are always welcome.

LaserGRBL is only an interface to communicate with GRBL firmware based systems, I simply use the COM port via the SerialPort object provided by the .NET framework

I don't use the driver directly, I don't deal with the driver development, nor with the CH340 or FTDI chip that I don't even know what it is. (joke, of course I know, but it's not my problem).

arkypita, 2020-03-19:
PS: I used a lot of kind of arduino clones and most of them works well, but some are buggy. Once I had a card sent by a user, because he had absurd behaviors that seemed inexplicable, and so I wanted to try to understand. I realized them that the quality of the clones is very diverse.

closed 2020-03-24 by arkypita