arkypita/LaserGRBL #927: Soft Limit / Ortur Laser

wuidling, 2020-03-19:
Hello,

I hava a problem with my new Ortur Laser..
Everythings fine wirh LaserGRBL if i make one run.

But if i want to add 2 or more runs the laser stops with the error Soft Limit.
I tried to change the streaming mode but only setting the Soft Limit to 0 solved the problem.

Any Ideas

Thanks

wuidling, 2020-03-19:
I find another solution in the forum.
https://github.com/arkypita/LaserGRBL/issues/916

I will try it tomorrow.

arkypita, 2020-03-19:
can you send me the gcode and your $$ configuration?

arkypita, 2020-03-19:
if it stop for "soft limit" it is because the gcode itself, or the code between passes, goes out of the configured machine boundary

wuidling, 2020-03-22:
Thanks for the quick response.
Here are some more informations:
[avrdude.txt](https://github.com/arkypita/LaserGRBL/files/4365181/avrdude.txt)
[image]
[gbrl.txt](https://github.com/arkypita/LaserGRBL/files/4365186/gbrl.txt)
Regards

wuidling, 2020-03-22:
G91
G0 Z-1 (sinks the Z axis)
G90

StuartB4, 2020-03-22:
You can probably disable Hard limits, change $21 to 0
Enable Soft Limits set $20 to 1
Make sure the exact engraving size is set correctly in $130 and 131 (Maximum Travel)
Also look in LaserGRBL settings on the GRBL menu and remove anything in the GCode boxes on the 5th tab across.

StuartB4, 2020-03-22:
If your home position is bottom left change $23 to 3

arkypita, 2020-03-22:
Shit. I added the code that lowers the Z axis (with relative mode) to help those who make more passes to automatically lower the laser, thinking that both those who did not have the Z axis the machine would have done no problem.
I didn't think that some could have a Z configuration that can generate alarms.

I know that in the next release I will change the default, going to comment on the line that lowers the Z axis (only for new installations)

arkypita, 2020-03-22:
@wuidling what is the value of $132 in ortur laser?

wuidling, 2020-03-23:
$132 1.000

I have deleted the entrys in multiple passes.

Regards

closed 2020-03-24 by arkypita