arkypita/LaserGRBL #928: Cannot connect to device. No such host is known.

Knight7171, 2020-03-21:
Hello everyone,

I purchased a Uttiny 6550 CNC laser engraver kit 10W on Amazon.
[https://www.amazon.com/gp/product/B0811TFNLD/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1](url)
I installed the CH340 driver and Installed the LaserGRBL software and updated it to v3.1.2. I also tested it on cardboard to make sure that everything works. And it did for almost 2 hours and now it does not. It tells my laptop (Windows 10 Pro) not connected.

Any help would be greatly appreciated.

Thank You!

[image]
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/4363167/sessionlog.txt)
[image]

gmmanonymus111, 2020-03-21:
From the first picture, I guess you accientally set it to Telnet connection instead of USB. To change it back, click `Grbl` -> `Settings` -> `Connection protocol`
[image]

Knight7171, 2020-03-21:
Thank You gmmanonymus111. I switched it back and no errors. I am not sure how it got changed but I am very happy that it is working again. And it was a easy fix. :) I have been trying to figure this out for 2 days. Once again Thank you!

closed 2020-03-24 by arkypita